ros2/rclpy #11: undo explicit blacklist in Python implementation selection

wjwwood, 2016-04-06:
Undo https://github.com/ros2/rclpy/pull/10.

Instead we should:

- Improve the error message when a Python extension doesn't exist for an otherwise valid rmw implementation
- Consider having a separate list of rmw implementations which provide a working Python extension
- Figure out how to de-duplicate the "default rmw implementation" logic that also exists in CMake: https://github.com/ros2/rmw/blob/739190b/rmw_implementation_cmake/cmake/get_default_rmw_implementation.cmake#L35

jacquelinekay, 2016-04-06:
Blacklist/whitelist-related issues to fix in rosidl_generator_py:

- [ ] Hardcoded map of rmw implementation to type support implementation (https://github.com/ros2/rosidl/blob/master/rosidl_generator_py/rosidl_generator_py/import_type_support_impl.py)
- [ ] Hardcoded list of supported typesupports in CMake (https://github.com/ros2/rosidl/blob/master/rosidl_generator_py/cmake/rosidl_generator_py_get_typesupports.cmake, with TODO comment in https://github.com/ros2/rosidl/blob/master/rosidl_generator_py/cmake/rosidl_generator_py_generate_interfaces.cmake)

jacquelinekay, 2016-05-02:
Is this ticket closeable? I believe recent changes using the ament_index have fixed the blacklisting problems in this repo.

wjwwood, 2016-05-04:
I guess so.

wjwwood, 2016-05-04:
Well, actually we haven't de-duplicated the logic for selecting the default rmw implementation. It's duplicated here:

https://github.com/ros2/rclpy/blob/master/rclpy/rclpy/impl/rmw_implementation_tools.py#L89

closed 2016-05-04 by wjwwood