ros2/rclpy #22: RCLPY_IMPLEMENTATION causes rclpytests to fail

jacquelinekay, 2016-05-03:
If I set the environment variable `RCLPY_IMPLEMENTATION` to a valid rmw implementation name and then run tests for rclpy, the set_rmw_implementation tests fail with output like:

```
1: Traceback (most recent call last):
1:   File "/usr/lib/python3.4/multiprocessing/pool.py", line 119, in worker
1:     result = (True, func(*args, **kwds))
1:   File "/home/jackie/code/ros2_ws/src/ros2/rclpy/rclpy/test/test_select_rmw_implementation.py", line 24, in run_catch_report_raise
1:     return func(*args, **kwargs)
1:   File "/home/jackie/code/ros2_ws/src/ros2/rclpy/rclpy/test/test_select_rmw_implementation.py", line 46, in func_import_each_available_rmw_implementation
1:     rmw_implementation, set_rmw)
1: AssertionError: expected 'rmw_connext_dynamic_cpp' but got 'rmw_connext_cpp'
1: """
1: 
1: The above exception was the direct cause of the following exception:
1: 
1: Traceback (most recent call last):
1:   File "/usr/lib/python3/dist-packages/nose/case.py", line 198, in runTest
1:     self.test(*self.arg)
1:   File "/home/jackie/code/ros2_ws/src/ros2/rclpy/rclpy/test/test_select_rmw_implementation.py", line 56, in test_import_each_available_rmw_implementation
1:     args=(func_import_each_available_rmw_implementation, rmw_implementation,)
1:   File "/usr/lib/python3.4/multiprocessing/pool.py", line 253, in apply
1:     return self.apply_async(func, args, kwds).get()
1:   File "/usr/lib/python3.4/multiprocessing/pool.py", line 599, in get
1:     raise self._value
1: AssertionError: expected 'rmw_connext_dynamic_cpp' but got 'rmw_connext_cpp'
```

The purposes of trying to was so that I could set a different rmw implementation for rclpytests to see if connext dynamic was working.

Perhaps the test should unset the variable before running. Additionally, the talker/listener test should be changed to iterate through the different rmw implementations.

mikaelarguedas, 2016-05-05:
@wjwwood I could take that off your plate if you want to

wjwwood, 2016-05-05:
Sure, it will be a few days until I get to it.

mikaelarguedas, 2016-05-09:
Consider rewriting these tests using independent files for the listener side to fix flaky/failing tests on buildfarm.

Is `rclpy.shutdown()` supposed to allow one to change the rmw_implementation or does that have to happen in a different process?
```python
import rclpy
from rclpy.impl.rmw_implementation_tools import get_rmw_implementations
from rclpy.impl.rmw_implementation_tools import select_rmw_implementation
select_rmw_implementation('rmw_opensplice_cpp')
rclpy.init()
#do stuff
rclpy.shutdown()
select_rmw_implementation('rmw_connext_cpp')
```
throws:
```python
raise ImplementationAlreadyImportedException()
```
Is that the expected behaviour ?

dirk-thomas, 2016-05-09:
I don't think it is currently possible. But this is an important use case often requested in ROS 1 which I think is important to support. The user should be able to `init`, run stuff and `shutdown` and then be able to start over again within the same process.

jacquelinekay, 2016-05-09:
this is a workaround, and I'm actually not sure if it would work, but: are global variables provided by modules cleaned up after an import goes out of scope?

for example, would this reset the rmw implementation singleton?

```
def test(rmw_implementation_name):
    import rclpy
    from rclpy.impl.rmw_implementation_tools import get_rmw_implementations
    from rclpy.impl.rmw_implementation_tools import select_rmw_implementation
    select_rmw_implementation(rmw_implementation_name)
    rclpy.init()
    #do stuff
    rclpy.shutdown()
```

The longterm solution would be to add code to shutdown to set global implementation-related state back to None (https://github.com/ros2/rclpy/blob/master/rclpy/rclpy/impl/rmw_implementation_tools.py#L25-L27).

wjwwood, 2016-05-09:
> I don't think it is currently possible. But this is an important use case often requested in ROS 1 which I think is important to support. The user should be able to init, run stuff and shutdown and then be able to start over again within the same process.

That is possible.

Being able to do:

- init
- shutdown
- init

Is different from:

- init
- shutdown
- change implementation
- init

The former should be possible in `rclpy` as-is. I don't think supporting the latter is extremely important though it would be nice if it worked. I do not believe it will though because I don't know how importing two Python extensions with the same symbols will work. If you would post the traceback, you'd see the error comes up when you try to change implementations after doing init, which is the issue here, not doing an init/shutdown/init cycle.

mikaelarguedas, 2016-05-09:
I don't think that the singleton gets reset on shutdown whether it's used
inside a function or globally.

Now that it's confirmed that this is a wanted feature, your solution seems
very suitable.
Thanks!

mikaelarguedas, 2016-05-09:
Yes indeed doing init -> shutdown -> init works with the current implementation. So we can currently shutdown and start again in the same process as long as you don't want to change implementation in between.

I don't see many use cases where people would want to change implementation like this. It's actually only useful for testing multiple rmw_implementations on our end. That's why I was wondering if I should just modify the tests to do this in separate processes or modify `rclpy` to make it possible.

jacquelinekay, 2016-05-09:
I don't think we need a test for it if it's not a feature we are going to support.

mikaelarguedas, 2016-05-09:
Sounds good to me.

wjwwood, 2016-05-09:
There are tests that test multiple implementations already by using the multiprocessing module. So you might look at those of you want to do it that way. But personally I think just having separate invocations of Python for each would be cleaner.

mikaelarguedas, 2016-05-09:
Yes I think that separating them is cleaner too. Thanks for the feedback

closed 2016-06-11 by mikaelarguedas