ros2/rclpy #31: Create custom exceptions in _rclpy.c

mikaelarguedas, 2016-05-27:
For now every failure in rclpy raises a `RuntimeError`, we should create custom errors for different failure scenarios and avoid the tests of just catching random RuntimeErrors.

suab321321, 2019-11-12:
@mikaelarguedas I would like to like to contribute

mikaelarguedas, 2019-11-28:
@suab321321 I haven't been involved in this project for a long time so can't provide feedback about the status / relevance of this ticket with respect to the current code base.
Maybe current maintainers will be able to provide more up-to-date information.

jacobperron, 2019-12-03:
It looks like we have a few custom exceptions defined already:

https://github.com/ros2/rclpy/blob/d375c84dadf00418da3823af8aa387d68fadb74a/rclpy/src/rclpy/_rclpy.c#L43-L45

But a quick search yields several places in [_rclpy.c](https://github.com/ros2/rclpy/blob/master/rclpy/src/rclpy/_rclpy.c) that raise `RuntimeError` where we might consider using custom exceptions. Whenever a call to the `rcl` layer errors, at the very least we could return a generic `RCLError` exception (I think this is what rclcpp does).

@suab321321 If you'd like to contribute to this ticket, I think a good first step would be to replace instances of RuntimeError with a new RCLError whenever there's a failed call to the rcl layer.
I suggest reviewing documentation on writing C-extensions for Python, in particular [this section on errors and exceptions](https://docs.python.org/3.6/extending/extending.html#intermezzo-errors-and-exceptions). You can add logic to initialize the new exception next to the other exceptions:

https://github.com/ros2/rclpy/blob/d375c84dadf00418da3823af8aa387d68fadb74a/rclpy/src/rclpy/_rclpy.c#L5088-L5099

suab321321, 2019-12-03:
@jacobperron okay sir I will proceed with this

suab321321, 2019-12-06:
@jacobperron sir I have to make RCLError first right?

jacobperron, 2019-12-06:
> I have to make RCLError first right?

Correct. You can declare and initialize it in the two places of the code I referenced (https://github.com/ros2/rclpy/issues/31#issuecomment-560948574).

suab321321, 2019-12-06:
#478 here @jacobperron sir

closed 2019-12-10 by jacobperron