ros2/rclpy #36: import messages before calling rclpy.init

mikaelarguedas, 2016-06-20:
modify typesupport import logic to allow message import before chosing rclpy implementation

mikaelarguedas, 2016-06-21:
@dirk-thomas do you think that this classmethod approach is a suitable solution to allow importing the typesupport at subscription/publication time ?

mikaelarguedas, 2016-06-22:
seems to have introduced [a single test failure](http://ci.ros2.org/job/ci_linux/1488/testReport/junit/(root)/test_publisher_subscriber_python__Primitives__rmw_opensplice_cpp_/test_publisher_subscriber/)

running repeated tests to see if the `proxy participant unknown` warning is related. 
[image]

mikaelarguedas, 2016-06-22:
Failure did not reappear.
Linux: [image]
OSX: [image]
Windows job failed while trying to delete the workspace before building. Will try again on windshield soon

mikaelarguedas, 2016-06-22:
@ros2/ros2_team please review

dirk-thomas, 2016-06-22:
+1 for the referenced PRs.

As discussed offline in the future this could be refactored to avoid that the Python code has to deal with it (check the class variables, call a function which populates them, and then pull the content of the variables in C again). Instead the C code should be able to access the information directly.

closed 2016-06-22 by mikaelarguedas