ros2/rclpy #42: Failed to take from a subscription: error not set

dhood, 2016-10-04:
listener_py is failing for me for all four RMW implementations. It will fail if a talker_py is not running. Even if a talker is running, it will also eventually fail, though not always immediately.

Here's the error:

```
$ listener_py
RTI Data Distribution Service EVAL License issued to SE dhood@osrfoundation.org For non-production use only.
Expires on 02-Nov-2016 See www.rti.com for more information.
I heard: [Hello World: 7]
Traceback (most recent call last):
  File "/Users/deanna/ros2_ws/install/bin/listener_py", line 9, in <module>
    load_entry_point('rclpy-examples', 'console_scripts', 'listener_py')()
  File "/Users/deanna/ros2_ws/build/rclpy_examples/listener_py.py", line 39, in main
    rclpy.spin_once(node)
  File "/Users/deanna/my_ros2_ws/install/lib/python3.5/site-packages/rclpy/__init__.py", line 72, in spin_once
    msg = _rclpy.rclpy_take(subscription.subscription_handle, subscription.msg_type)
RuntimeError: Failed to take from a subscription: error not set
```

dirk-thomas, 2016-10-04:
They all worked for me on Linux built from source against the professional version of Connext.

wjwwood, 2016-10-04:
I also don't see this issue on my mac using the binary from ci.ros2.org.

dhood, 2016-10-04:
I don't see the issue from the binary either, @wjwwood. Do you see it from source?

wjwwood, 2016-10-04:
I haven't tried, let me build it really quick.

wjwwood, 2016-10-04:
I still don't see it after building from source, I think @dhood is going to dig a bit deeper with print statements in the C code.

dhood, 2016-10-05:
Restarting my machine and rebuilding from scratch fixed it.

closed 2016-10-05 by dhood