ros2/rclpy #68: Get current time?

sidd1, 2017-02-23:
Is there an equivalent for `rospy.Time.now()` with rclpy? I know I can use the time function and build my own `Time` message for the standard `Header` messages. I want to know if this has already been implemented?

mikaelarguedas, 2017-02-23:
Unfortunately it is not implemented yet, there is a ticket [tracking this](https://github.com/ros2/ros2/issues/178)
For now the [design](http://design.ros2.org/articles/clock_and_time.html) and the API have been defined but the feature itself has not been implemented yet.

sidd1, 2017-02-23:
Ah yes I see. I read the design document but could not find an issue here or at `rclcpp`, which threw me off. Thanks! I will close this issue then.

closed 2017-02-23 by sidd1

stonier, 2018-10-30:
For reference, the breadcrumb trail for rclpy's version leads to #186