ros2/rclpy #69: KeyboardInterrupt not being raised out of publish

wjwwood, 2017-02-24:
You can reproduce this sometimes with a simple Python script like this one:

```python
#!/usr/bin/env python3

import sys

import rclpy
from rclpy.qos import qos_profile_default

from sensor_msgs.msg import Imu

def main():
    rclpy.init(sys.argv)

    node = rclpy.create_node('imu_publisher')

    imu_publisher = node.create_publisher(Imu, 'imu_data', qos_profile_default)

    imu_msg = Imu()

    while True:
        # busy publish
        imu_publisher.publish(imu_msg)

if __name__ == '__main__':
    sys.exit(main())

```

Since this code spends most of the time in the `.publish()` function, which is mostly implemented inside a C python extension, you can sometimes get a segfault when doing `ctrl-C` rather than the expected `KeyboardInterrupt`. Note that I'm not using spin here.

For example, I get this from a similarly structured script:

```
william@ubuntu ~/ros2_ws 
% python3 emulate_kobuki_node.py
^C[1]    26835 segmentation fault (core dumped)  python3 emulate_kobuki_node.py

william@ubuntu ~/ros2_ws 
% python3 emulate_kobuki_node.py
^C[1]    26859 segmentation fault (core dumped)  python3 emulate_kobuki_node.py

william@ubuntu ~/ros2_ws 
% python3 emulate_kobuki_node.py
^C[1]    26883 segmentation fault (core dumped)  python3 emulate_kobuki_node.py

william@ubuntu ~/ros2_ws 
% python3 emulate_kobuki_node.py
^CTraceback (most recent call last):
  File "emulate_kobuki_node.py", line 39, in <module>
    sys.exit(main())
  File "emulate_kobuki_node.py", line 29, in main
    odom_publisher.publish(odom_msg)
  File "/home/william/ros2_ws/install_isolated/rclpy/lib/python3.5/site-packages/rclpy/publisher.py", line 27, in publish
    _rclpy.rclpy_publish(self.publisher_handle, msg)
  File "/home/william/ros2_ws/install_isolated/rclpy/lib/python3.5/site-packages/rclpy/impl/object_proxy.py", line 32, in __getattribute__
    def __getattribute__(self, attr):
KeyboardInterrupt
```

mikaelarguedas, 2017-08-08:
indeed, keyboard interrupts are caught only in `rclpy_wait` for now. Good thing to keep in mind in the upcoming refactor.

I updated the code snippet to match the new api

dirk-thomas, 2017-10-17:
Please try the patch from ros2/rosidl#241 and report if it fixes the problem for you too.

mikaelarguedas, 2017-10-17:
I couldn't reproduce the segfault when using the patch in ros2/rosidl#241 :+1:

closed 2017-10-18 by dirk-thomas