ros2/rclpy #83: Error management: replace asserts where appropriate

mikaelarguedas, 2017-05-10:
Right now [_rclpy.c](https://github.com/ros2/rclpy/blob/master/rclpy/src/rclpy/_rclpy.c) is full of asserts. Would be great to throw or return accordingly. This is also related to https://github.com/ros2/rclpy/issues/31

allenh1, 2017-07-12:
@mikaelarguedas is anyone working on this?

mikaelarguedas, 2017-07-12:
This will be part of the rclpy refactor that will be discussed next week along with #95 #31 #58 so I'd rather wait to do it in a consistent way to be sure not to conflict.

#84 is still available for grabs though

Ridhwanluthra, 2017-09-19:
is someone working on this?
I can get this done. Please provide some more details
Thanks

clalancette, 2024-12-30:
Since we moved away from `_rclpy.c`, I'm going to close this one out.

closed 2024-12-30 by clalancette