ros2/rclpy #103: Expose logging macros in rclpy

dhood, 2017-08-22:
meta-ticket.

First question: which usage do we want with respect to the severity:

 1. `loginfo(...)` (like in ROS 1)
 1. `log(..., severity=INFO)`

Second question: when considering the other logging variations, which do we want?

 1. `log_info_throttle_named(...)` (like the C usage)
 1. `log_info(..., throttle_duration=1.0, name='my_name')` (more pythonic IMO)

dirk-thomas, 2017-08-22:
We talked about this in the meeting:

Regarding the first question: use 2. `log(..., severity=INFO)` but offer 1. `loginfo(...)` for convenience.

Regarding the second question: the keyword argument (2.) is preferred.

Additionally it would be nice to have a `get_named_logger(name='xxx')` which can be called once and then be passed around. That logger could have `log(..., severity=INFO)` as well as `info(...)` for convenience.

mikaelarguedas, 2017-09-08:
closing this now that #102 has been merged

closed 2017-09-08 by mikaelarguedas