ros2/rclpy #115: r2b3 Failure to import _rclpy on windows

sloretz, 2017-09-12:
## Bug report

**Required Info:**

- Operating System:
  - Windows 10 (Fresh install on a VM)
- Installation type:
  - from archive http://ci.ros2.org/view/packaging/job/packaging_windows/728/artifact/ws/ros2-package-windows-AMD64.zip
- Version or commit hash:
  - def304b4270b6b935b7d6fbbf7671634c2532452
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue
Symptom is a lot of errors when using `ros2` command. This is after `call C:\dev\ros2\local_setup.bat`
```
C:\Users\osrf>ros2 --help
Failed to load entry point 'start': DLL load failed: The specified module could not be found.
Failed to load entry point 'status': DLL load failed: The specified module could not be found.
Failed to load entry point 'stop': DLL load failed: The specified module could not be found.
Failed to load entry point 'list': DLL load failed: The specified module could not be found.
Failed to load entry point 'call': DLL load failed: The specified module could not be found.
Failed to load entry point 'list': DLL load failed: The specified module could not be found.
Failed to load entry point 'echo': DLL load failed: The specified module could not be found.
Failed to load entry point 'list': DLL load failed: The specified module could not be found.
Failed to load entry point 'pub': DLL load failed: The specified module could not be found.
usage: ros2 [-h] Call `ros2 <command> -h` for more detailed usage. ...

ros2 is an extensible command-line tool for ROS 2.

optional arguments:
  -h, --help            show this help message and exit

Commands:
  daemon    Various daemon related sub-commands
  msg       Various msg related sub-commands
  node      Various node related sub-commands
  pkg       Various package related sub-commands
  run       Run a package specific executable
  security  Various security related sub-commands
  service   Various service related sub-commands
  srv       Various srv related sub-commands
  topic     Various topic related sub-commands

  Call `ros2 <command> -h` for more detailed usage.
```

Trying to load one of the entry points shows a traceback when importing _rclpy

```
>>> from ros2cli.verb.daemon import start
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\dev\ros2\Lib\site-packages\ros2cli\verb\daemon\start.py", line 15, in <module>
    from ros2cli.node.daemon import is_daemon_running
  File "C:\dev\ros2\Lib\site-packages\ros2cli\node\daemon.py", line 22, in <module>
    import rclpy
  File "C:\dev\ros2\Lib\site-packages\rclpy\__init__.py", line 17, in <module>
    from rclpy.executors import SingleThreadedExecutor as _SingleThreadedExecutor
  File "C:\dev\ros2\Lib\site-packages\rclpy\executors.py", line 21, in <module>
    from rclpy.impl.implementation_singleton import rclpy_implementation as _rclpy
  File "C:\dev\ros2\Lib\site-packages\rclpy\impl\implementation_singleton.py", line 32, in <module>
    rclpy_implementation = importlib.import_module('._rclpy', package='rclpy')
  File "C:\Python36\lib\importlib\__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ImportError: DLL load failed: The specified module could not be found.
```

Trying to import rclpy shows a similar error
```
>>> import rclpy
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\dev\ros2\Lib\site-packages\rclpy\__init__.py", line 17, in <module>
    from rclpy.executors import SingleThreadedExecutor as _SingleThreadedExecutor
  File "C:\dev\ros2\Lib\site-packages\rclpy\executors.py", line 21, in <module>
    from rclpy.impl.implementation_singleton import rclpy_implementation as _rclpy
  File "C:\dev\ros2\Lib\site-packages\rclpy\impl\implementation_singleton.py", line 32, in <module>
    rclpy_implementation = importlib.import_module('._rclpy', package='rclpy')
  File "C:\Python36\lib\importlib\__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ImportError: DLL load failed: The specified module could not be found.
```

#### Expected behavior
I'd expect no ImportError to be raised.

#### Actual behavior
Import error is raised

#### Additional information
sys.path
```
>>> import sys
>>> sys.path
['', 'C:\\dev\\ros2\\Lib\\site-packages', 'C:\\Python36\\python36.zip', 'C:\\Python36\\DLLs', 'C:\\Python36\\lib', 'C:\\Python36', 'C:\\Python36\\lib\\site-packages']
```

Not sure if I should see _rclpy.dll or something like that, but I don't.
```
C:\Users\osrf>dir C:\dev\ros2\Lib\site-packages\rclpy
 Volume in drive C has no label.
 Volume Serial Number is AA39-7FB6

 Directory of C:\dev\ros2\Lib\site-packages\rclpy

09/12/2017  09:38 AM    <DIR>          .
09/12/2017  09:38 AM    <DIR>          ..
09/12/2017  09:38 AM             3,527 callback_groups.py
09/12/2017  09:38 AM             3,126 client.py
09/12/2017  09:38 AM               646 constants.py
09/12/2017  09:38 AM             2,558 exceptions.py
09/12/2017  09:38 AM            16,545 executors.py
09/12/2017  09:38 AM             1,505 expand_topic_name.py
09/12/2017  09:38 AM    <DIR>          impl
09/12/2017  09:38 AM             2,535 logging.py
09/12/2017  09:38 AM            12,211 node.py
09/12/2017  09:38 AM             1,087 publisher.py
09/12/2017  09:38 AM             5,581 qos.py
09/12/2017  09:38 AM             1,449 service.py
09/12/2017  09:38 AM             1,270 subscription.py
09/12/2017  09:38 AM             2,157 timer.py
09/12/2017  09:38 AM             1,210 topic_or_service_is_hidden.py
09/12/2017  09:38 AM             1,203 utilities.py
09/12/2017  09:38 AM             1,709 validate_full_topic_name.py
09/12/2017  09:38 AM             1,568 validate_namespace.py
09/12/2017  09:38 AM             1,378 validate_node_name.py
09/12/2017  09:38 AM             1,710 validate_topic_name.py
09/12/2017  09:38 AM             1,716 _rclpy.lib
09/12/2017  09:38 AM            44,032 _rclpy.pyd
09/12/2017  09:38 AM             1,932 _rclpy_logging.lib
09/12/2017  09:38 AM            11,264 _rclpy_logging.pyd
09/12/2017  09:38 AM             1,654 __init__.py
09/12/2017  09:46 AM    <DIR>          __pycache__
              24 File(s)        123,573 bytes
               4 Dir(s)   6,535,761,920 bytes free
```

dirk-thomas, 2017-09-12:
Same as https://discourse.ros.org/t/ros2-debugmodulenotfounderror-no-module-named-rclpy--rclpy/2401

sloretz, 2017-09-12:
Possibly due to missing VS2015 redistributable libraries? [Dependency Walker](http://www.dependencywalker.com/) reports these as missing

```
API-MS-WIN-CORE-APIQUERY-L1-1-0.DLL
API-MS-WIN-CORE-APPCOMPAT-L1-1-1.DLL
API-MS-WIN-CORE-APPINIT-L1-1-0.DLL
API-MS-WIN-CORE-ATOMS-L1-1-0.DLL
API-MS-WIN-CORE-COMM-L1-1-0.DLL
API-MS-WIN-CORE-CONSOLE-L1-1-0.DLL
API-MS-WIN-CORE-CONSOLE-L2-1-0.DLL
API-MS-WIN-CORE-CRT-L1-1-0.DLL
API-MS-WIN-CORE-CRT-L2-1-0.DLL
API-MS-WIN-CORE-DATETIME-L1-1-1.DLL
API-MS-WIN-CORE-DATETIME-L1-1-2.DLL
API-MS-WIN-CORE-DEBUG-L1-1-1.DLL
API-MS-WIN-CORE-DELAYLOAD-L1-1-1.DLL
API-MS-WIN-CORE-ERRORHANDLING-L1-1-1.DLL
API-MS-WIN-CORE-ERRORHANDLING-L1-1-3.DLL
API-MS-WIN-CORE-FIBERS-L1-1-1.DLL
API-MS-WIN-CORE-FIBERS-L2-1-1.DLL
API-MS-WIN-CORE-FILE-L1-2-1.DLL
API-MS-WIN-CORE-FILE-L1-2-2.DLL
API-MS-WIN-CORE-FILE-L2-1-1.DLL
API-MS-WIN-CORE-FILE-L2-1-2.DLL
API-MS-WIN-CORE-HANDLE-L1-1-0.DLL
API-MS-WIN-CORE-HEAP-L1-2-0.DLL
API-MS-WIN-CORE-HEAP-L2-1-0.DLL
API-MS-WIN-CORE-HEAP-OBSOLETE-L1-1-0.DLL
API-MS-WIN-CORE-INTERLOCKED-L1-2-0.DLL
API-MS-WIN-CORE-IO-L1-1-1.DLL
API-MS-WIN-CORE-JOB-L1-1-0.DLL
API-MS-WIN-CORE-KERNEL32-LEGACY-L1-1-1.DLL
API-MS-WIN-CORE-KERNEL32-PRIVATE-L1-1-1.DLL
API-MS-WIN-CORE-LARGEINTEGER-L1-1-0.DLL
API-MS-WIN-CORE-LIBRARYLOADER-L1-2-0.DLL
API-MS-WIN-CORE-LIBRARYLOADER-L1-2-1.DLL
API-MS-WIN-CORE-LIBRARYLOADER-L2-1-0.DLL
API-MS-WIN-CORE-LOCALIZATION-L1-2-1.DLL
API-MS-WIN-CORE-LOCALIZATION-L2-1-0.DLL
API-MS-WIN-CORE-LOCALIZATION-OBSOLETE-L1-3-0.DLL
API-MS-WIN-CORE-LOCALIZATION-PRIVATE-L1-1-0.DLL
API-MS-WIN-CORE-MEMORY-L1-1-2.DLL
API-MS-WIN-CORE-NAMEDPIPE-L1-2-0.DLL
API-MS-WIN-CORE-NAMEDPIPE-L1-2-1.DLL
API-MS-WIN-CORE-NAMESPACE-L1-1-0.DLL
API-MS-WIN-CORE-NORMALIZATION-L1-1-0.DLL
API-MS-WIN-CORE-PATH-L1-1-0.DLL
API-MS-WIN-CORE-PERFCOUNTERS-L1-1-0.DLL
API-MS-WIN-CORE-PRIVATEPROFILE-L1-1-1.DLL
API-MS-WIN-CORE-PROCESSENVIRONMENT-L1-2-0.DLL
API-MS-WIN-CORE-PROCESSSNAPSHOT-L1-1-0.DLL
API-MS-WIN-CORE-PROCESSTHREADS-L1-1-2.DLL
API-MS-WIN-CORE-PROCESSTHREADS-L1-1-3.DLL
API-MS-WIN-CORE-PROCESSTOPOLOGY-L1-2-0.DLL
API-MS-WIN-CORE-PROFILE-L1-1-0.DLL
API-MS-WIN-CORE-PSAPI-ANSI-L1-1-0.DLL
API-MS-WIN-CORE-PSAPI-L1-1-0.DLL
API-MS-WIN-CORE-QUIRKS-L1-1-0.DLL
API-MS-WIN-CORE-REALTIME-L1-1-0.DLL
API-MS-WIN-CORE-REGISTRY-L1-1-0.DLL
API-MS-WIN-CORE-REGISTRY-L1-1-1.DLL
API-MS-WIN-CORE-REGISTRYUSERSPECIFIC-L1-1-0.DLL
API-MS-WIN-CORE-RTLSUPPORT-L1-2-0.DLL
API-MS-WIN-CORE-SHLWAPI-LEGACY-L1-1-0.DLL
API-MS-WIN-CORE-SHLWAPI-OBSOLETE-L1-2-0.DLL
API-MS-WIN-CORE-SIDEBYSIDE-L1-1-0.DLL
API-MS-WIN-CORE-STRING-L1-1-0.DLL
API-MS-WIN-CORE-STRING-L2-1-0.DLL
API-MS-WIN-CORE-STRING-L2-1-1.DLL
API-MS-WIN-CORE-STRING-OBSOLETE-L1-1-0.DLL
[... 291 lines cut ...]
EXT-MS-WIN-RTCORE-NTUSER-WINDOW-EXT-L1-1-0.DLL
EXT-MS-WIN-SECUR32-TRANSLATENAME-L1-1-0.DLL
EXT-MS-WIN-SECURITY-CAPAUTHZ-L1-1-0.DLL
EXT-MS-WIN-SECURITY-CHAMBERS-L1-1-0.DLL
EXT-MS-WIN-SECURITY-CREDUI-L1-1-0.DLL
EXT-MS-WIN-SECURITY-EFS-L1-1-0.DLL
EXT-MS-WIN-SESSION-USERMGR-L1-1-0.DLL
EXT-MS-WIN-SESSION-USERTOKEN-L1-1-0.DLL
EXT-MS-WIN-SESSION-WINSTA-L1-1-0.DLL
EXT-MS-WIN-SESSION-WTSAPI32-L1-1-0.DLL
EXT-MS-WIN-SETUPAPI-INF-L1-1-0.DLL
EXT-MS-WIN-SHELL-DIRECTORY-L1-1-0.DLL
EXT-MS-WIN-SHELL-PROPSYS-L1-1-0.DLL
EXT-MS-WIN-SHELL-SHELL32-L1-2-0.DLL
EXT-MS-WIN-SHELL-SHLWAPI-L1-1-0.DLL
EXT-MS-WIN-SHELL32-SHELLCOM-L1-1-0.DLL
EXT-MS-WIN-SXS-OLEAUTOMATION-L1-1-0.DLL
EXT-MS-WIN-UI-VIEWMANAGEMENT-L1-1-0.DLL
EXT-MS-WIN-USP10-L1-1-0.DLL
EXT-MS-WIN-WER-UI-L1-1-0.DLL
EXT-MS-WIN-WER-XBOX-L1-1-0.DLL
EXT-MS-WIN-WINRT-STORAGE-L1-1-0.DLL
EXT-MS-WIN-WLAN-ONEXUI-L1-1-0.DLL
IESHIMS.DLL
```

mikaelarguedas, 2017-09-12:
also similar to https://github.com/ros2/ros2/issues/360#issuecomment-316492672

mikaelarguedas, 2017-09-12:
Installing openssl fixed the problem, added to the installation instructions in https://github.com/ros2/ros2/wiki/Windows-Install-Binary/_compare/3bb08f69ccbba3841af5b481d5263815165ae314...1089afdf20f981940c50a222e96f4c6a591b6de2.

The issue is that since beta2 we ship fastrtps with security. So we need some openssl dlls to be able to run our programs. The error message given by python is not clear at all. I tracked it down when trying to run c++ code that gave a much clearer like: "libEAY32.dll is missing".

Installing openSSL fixed both c++ and Python examples

dirk-thomas, 2017-09-12:
Can you look into the raised `ImportError` and what it's members contain (`msg`, `name`, `path`). Maybe there is some useful information in there?

dhood, 2017-09-12:
this might explain users having issues with OSX binaries since we shipped with security, too

sloretz, 2017-09-12:
@dirk-thomas Here are the members, haven't tried installing openssl yet

```
>>> try:
...    import rclpy
... except Exception as e:
...    print(dir(e))
...    print ("args", e.args, "msg", e.msg, "name", e.name, "path", e.path)
...
['__cause__', '__class__', '__context__', '__delattr__', '__dict__', '__dir__', '__doc__', '__eq__', '__format__', '__ge__', '__getattribute__', '__gt__', '__hash__', '__init__', '__init_subclass__', '__le__', '__lt__', '__ne__', '__new__', '__reduce__', '__reduce_ex__', '__repr__', '__setattr__', '__setstate__', '__sizeof__', '__str__', '__subclasshook__', '__suppress_context__', '__traceback__', 'args', 'msg', 'name', 'path', 'with_traceback']
args ('DLL load failed: The specified module could not be found.',) msg DLL load failed: The specified module could not be found. name _rclpy path C:\dev\ros2\Lib\site-packages\rclpy\_rclpy.pyd
```

mikaelarguedas, 2017-09-12:
Thanks @sloretz for posting this. I can't reproduce it anymore... Removing env variable and openssl DLLs from the path is not enough. Uninstalling OpenSSL is not enough either o_O. Not sure how to get back in the broken state...

dirk-thomas, 2017-09-12:
Since `e.path` points to the loaded module we could add some logic in case of this exception which checks that this file actually exists and if it does print a message which indicates that dependencies of that library are likely missing and optionally include a guess what the user might need / point to a wiki page providing a detailed description.

dirk-thomas, 2017-09-12:
I would keep the logic simple and point to a single page / ticket which than provides enough information to guide the user further.

mikaelarguedas, 2017-09-12:
> Uninstalling OpenSSL is not enough either o_O. Not sure how to get back in the broken state...

Follow up on this: looks like uninstalling openssl didnt remove the culprit dll from System32 that's why it's still working. Removing it by hand allows me to reproduce

sloretz, 2017-09-12:
@dirk-thomas did you mean that ros2cli would print the warning, or that it would be printed from here? https://github.com/ros2/rclpy/blob/1ef2924ef8e154c0553edf0fdba4840b08b728f8/rclpy/rclpy/impl/implementation_singleton.py#L32

dirk-thomas, 2017-09-12:
It needs to be handled in `rclpy` wrapping the line you referenced.

PR #116 (2017-09-12): add hint if rclpy fails to be imported
closes #115 
connects to #115 

Thew content of the [referenced wiki page](https://github.com/ros2/ros2/wiki/Rclpy-Import-error-hint) can be updated at a later date

Without this change
```
Failed to load entry point 'pub': DLL load failed: The specified module could not be found.
Traceback (most recent call last):
  File "C:\dev\ros2\lib\demo_nodes_py\talker-script.py", line 11, in <module>
    load_entry_point('demo-nodes-py==0.0.0', 'console_scripts', 'talker')()
  File "C:\Python36\lib\site-packages\pkg_resources\__init__.py", line 570, in load_entry_ [... cut]
closed 2017-09-12 by mikaelarguedas