ros2/rclpy #124: Use PyCapsule name to check capsule pointer types

sloretz, 2017-10-10:
## Feature request

#### Feature description
Currently rclpy creates `PyCapsule` with NULL names. This means if a function is given a `PyCapsule` containing `rcl_client_t *` but it expects `rcl_node_t *` the C code will still attempt to use it as such. If the capsules were given a name at creation then the methods could use the name as a type check when getting the pointer.

#### Implementation considerations
https://docs.python.org/3/c-api/capsule.html
`PyCapsule_New` accepts a `const char *name` argument. It could be given a string like `"rcl_node_t *"`. 

```C
// rcl_node_t * node;
PyObject * pynode = PyCapsule_New(node, "rcl_node_t *");
```

 `PyCapsule_GetPointer` accepts a `const char *name` argument, and it sets a python exception and returns `NULL` if the name does not match (it uses `strcmp()` ).

```C
rcl_node_t * node = (rcl_node_t *)PyCapsule_GetPointer(pynode, "rcl_node_t *");
if (!node)
{
  // GetPointer already set an exception
  return NULL;
}
```

If the default exception isn't appropriate, `PyCapsule_GetPointer` is guaranteed to succeed if `PyCapsule_IsValid` returns true. This can be checked first with custom handling code to run if it returns False.

```C
// PyObject * pynode;
if (!PyCapsule_IsValid(pynode, "rcl_node_t *"))
{
  // Do something about it
}
```

mikaelarguedas, 2017-10-10:
:+1: we discussed it but never got around renaming all the capsules

mikaelarguedas, 2017-10-26:
that's totally done and working now

closed 2017-10-26 by mikaelarguedas