ros2/rclpy #126: rclpy beta3 requires libopensplice67 to be installed

BrannonKing, 2017-10-11:
## Bug report

**Required Info:**

- Operating System:
Ubuntu 16.04
- Installation type:
Custom build including CoreDX, FastRTPS, and OpenSplice
- Version or commit hash:
Beta3 (exactly)

#### Steps to reproduce issue
Start a python3 ROS2b3 node without libopensplice67 installed. Send it some data or have it send some.

#### Expected behavior
No problems with libopensplice67 absent -- just like the rclcpp nodes.

#### Actual behavior
You get this error and crash: ros2-linux/lib/libstd_msgs__rosidl_typesupport_opensplice_cpp.so: error: symbol lookup error: undefined symbol: _ZTIN3DDS10OpenSplice21TypeSupportMetaHolderE (fatal)

I've attached the full LD log.
[ld_debug.txt.zip](https://github.com/ros2/rclpy/files/1377468/ld_debug.txt.zip)

dirk-thomas, 2017-10-12:
Can you please clarify your scenario a bit more. In the "required info" section you mention that you built with OpenSplice but in the "steps to reproduce" you say `libopensplice67` isn't installed. Did you uninstall it between building and using the workspace?

BrannonKing, 2017-10-12:
My build machine is not the same machine as the machines that I deploy onto. I can make them the same; I didn't want to have to do that. I also use different middleware in different deployments; it's nice to have a master build that supports them all, but I don't have a lot of extra hardware resources on some platforms.

BrannonKing, 2017-10-12:
Yes, this is a duplicate of the issue 399 referenced. I thought that was already a feature.

dirk-thomas, 2017-11-10:
Addressed by the PRs referenced in ros2/ros2#399.

closed 2017-11-10 by dirk-thomas