ros2/rclpy #136: Negative ref count in test_node.py on windows debug

dhood, 2017-10-31:
`test_create_service` has been giving errors sometimes on windows debug nightlies lately (first time Oct 28):

```
Fatal Python error: C:\J\workspace\nightly_win_deb\ws\src\ros2\rclpy\rclpy\src\rclpy\_rclpy.c:1444 object at 000001F7B40882C0 has negative ref count -7870023856155525121
ValueError: Failed to create service due to invalid topic name '/get/42parameters': topic name is invalid, at C:\J\workspace\nightly_win_deb\ws\src\ros2\rcl\rcl\src\rcl\expand_topic_name.c:67

Current thread 0x00000a38 (most recent call first):
  File "C:/J/workspace/nightly_win_deb/ws/src/ros2/rclpy/rclpy\rclpy\node.py", line 177 in create_service
  File "C:\J\workspace\nightly_win_deb\ws\src\ros2\rclpy\rclpy\test\test_node.py", line 75 in test_create_service
```

http://ci.ros2.org/job/nightly_win_deb/659/testReport/junit/(root)/projectroot/rclpytests/

It seems to be unhappy with this `Py_DECREF` upon service init failure that was added in https://github.com/ros2/rclpy/pull/129:
https://github.com/ros2/rclpy/pull/129/files#diff-80dda03110b5be606b823d6b0558b5c2R1444

Looking closer it seems to just be accidentally referencing `service` instead of `pyservice` so I'll fix it up now (still opening this ticket for future reference and visibility).

sloretz, 2017-10-31:
Oops, thanks for fixing it!

dhood, 2017-10-31:
not a problem, https://github.com/ros2/rclpy/pull/137

PR #137 (2017-10-31): Fix incorrect reference to service instead of pyservice
Fixes #136 

Standard CI:
* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

Windows Debug retest-until-fail 10 on rclpy: [image]
closed 2017-10-31 by dhood