ros2/rclpy #139: Check return value of Python functions that can fail

sloretz, 2017-11-01:
This is a tracking ticket for checking return values in rclpy. These functions are used and can fail, but their return value is not being checked.

- [ ] `PyList_New`
- [ ] `PyCapsule_New`
- [ ] `PyLong_FromUnsignedLongLong`
- [ ] `PyMem_Malloc`
- [ ] `PyUnicode_FromString`
- [ ] `PyLong_FromSize_t`
- [ ] `PyObject_GetAtterString`
- [ ] `PyLong_FromLongLong`
- [ ] `PyCapsule_GetName`
- [ ] `PyList_Append`
- [ ] `PyObject_CallObject`
- [ ] `PyTuple_New`
- [ ] `PyTuple_SetItem`
- [ ] `PyList_SetItem`
- [ ] `PyImport_ImportModule`
- [ ] `PyObject_SetAttrString`

From discussion here
https://github.com/ros2/rclpy/pull/138#discussion_r148367820
https://github.com/ros2/rclpy/pull/138#discussion_r148363958

db118, 2018-08-18:
First time contributing to open source and interested in learning more about how to fix this issue.

sloretz, 2018-09-06:
Hi @db118 ,

This issue is about checking error codes of python functions. Here is an example of error checking done correctly: `PyArg_ParseTuple(...)` can fail, and if it does the function should return `NULL` right away.

https://github.com/ros2/rclpy/blob/62012d3650de790d89c725232451c4e998b396fb/rclpy/src/rclpy/_rclpy.c#L1550-L1552

Here is an example of what should be fixed: `PyList_New()` can set an exception and return `NULL`, so `pylist` needs to be checked before calling `PyList_SET_ITEM()` with it.

https://github.com/ros2/rclpy/blob/62012d3650de790d89c725232451c4e998b396fb/rclpy/src/rclpy/_rclpy.c#L1693-L1697

A good first PR would fix one instance of a function that can fail not being checked.

db118, 2018-09-07:
@sloretz Ahhh that make sense 😄

db118, 2018-09-07:
@sloretz 
Do you want me to switch this null error checking 
https://github.com/ros2/rclpy/blob/62012d3650de790d89c725232451c4e998b396fb/rclpy/src/rclpy/_rclpy.c#L3642-L3645

to this implementation instead
https://github.com/ros2/rclpy/blob/62012d3650de790d89c725232451c4e998b396fb/rclpy/src/rclpy/_rclpy.c#L1550-L1552

sloretz, 2018-09-10:
@db118 See the section ["Building from source"](https://github.com/ros2/ros2/wiki/Installation) for instructions to build this library. It lists the dependencies, as well as shows how to build the other ros 2 packages this one depends on. Once you've built everything once, you can build and test just this backage with `colcon build --packages-select rclpy` and `colcon test --event-handlers console_direct+ --packages-select rclpy`.

I don't think I understand the question about swtiching `NULL` error checking; both of those look fine to me.

suab321321, 2019-06-23:
I would like to help with this one

suab321321, 2019-06-23:
Can you tell me in which file I need to fix this?

pr-db, 2021-08-12:
Hello, I'm new to open source and love working with ros. How can I approach this issue?

clalancette, 2021-08-12:
Given our recent switch to pybind11 (which gets completed with https://github.com/ros2/rclpy/pull/814), I actually think this issue is mostly obsolete.  @sloretz what do you think?  Should we just close this one out?

closed 2021-08-12 by sloretz