ros2/rclpy #142: Reset error message when rcl_take_response failed

minggangw, 2017-11-13:
I noticed that the implement of [`rcl_take_response`](https://github.com/ros2/rcl/blob/master/rcl/src/rcl/client.c#L258) in `rcl` returns **RCL_RET_ERROR** and **RCL_RET_CLIENT_TAKE_FAILED** when the execution failed. So when this circumstance occurs, we should reset the error message to avoid being overwritten later. I have submitted a pr #141, PTAL thanks!

dhood, 2017-11-14:
This is now essentially a duplicate of ros2/rclcpp#397 so I will close this issue. As mentioned in https://github.com/ros2/rclpy/pull/141#issuecomment-344032668, if you were seeing unexpected error output then https://github.com/ros2/rclpy/pull/143 may have fixed it for you.

closed 2017-11-14 by dhood

minggangw, 2017-11-14:
Thanks @dhood , that's reasonable.