ros2/rclpy #156: Can't create classes inheriting from node without explicitly importing rclpy.node.Node

mikaelarguedas, 2017-11-27:
To settle on before tagging the release.

Since #147, all classes inheriting from `rclpy.Node` need to explicitly `import rclpy.node.Node` (see https://github.com/ros2/demos/pull/196).

It would be great to allow users to use that class without having to explicitly import it as it was the case before.

@dirk-thomas could you provide more information about the need of delaying importing the singleton and the possible alternatives to restore previous usability?

dirk-thomas, 2017-12-03:
> could you provide more information about the need of delaying importing the singleton and the possible alternatives to restore previous usability?

The import of the singleton needs to be delayed in order to allow to override the logic where it should be loaded from.

The Python module `rclpy.node` would need to be importable *without* importing `rclpy.impl.implementation_singleton`. Then it could be imported in the `rclpy/__init__.py` to make it accessible as before.

dhood, 2017-12-06:
As a note: `rclpy.logging` is not being imported automatically either (it could be imported manually but it also imports a C extension and FWIU that's to be avoided): https://github.com/ros2/demos/pull/210