ros2/rclpy #178: occasional segfault in shutdown, perhaps in node.destroy_node()

wjwwood, 2018-03-03:
While testing some things for a discourse question (https://discourse.ros.org/t/how-to-shutdown-and-reinitialize-a-publisher-node-in-ros-2/4090/3?u=wjwwood), I created this script:

```python
import sys

import rclpy
from rclpy.node import Node

from std_msgs.msg import String

def main():
    msg = String()
    i = 0

    while True:
        rclpy.init()

        node = Node('talker')
        pub = node.create_publisher(String, 'chatter')
        msg.data = 'Hello World: {0}'.format(i)
        i += 1
        node.get_logger().info('Publishing: "{0}"'.format(msg.data))
        pub.publish(msg)

        node.destroy_node()
        rclpy.shutdown()

if __name__ == '__main__':
    main()
```

Which works fine, but one time when I did ctrl-c on it I got a segmentation fault:

```
...
[INFO] [talker]: Publishing: "Hello World: 336"
[INFO] [talker]: Publishing: "Hello World: 337"
^CTraceback (most recent call last):
  File "/tmp/testing/talker.py", line 31, in <module>
    main()
  File "/tmp/testing/talker.py", line 26, in main
    node.destroy_node()
  File "/Users/william/ros2_ws/install_release/lib/python3.6/site-packages/rclpy/node.py", line 286, in destroy_node
    _rclpy.rclpy_destroy_entity(self.handle)
KeyboardInterrupt
[1]    48020 segmentation fault  python3 /tmp/testing/talker.py
```

I've been trying to catch it in the debugger but I haven't been able to do so, and I don't really have time to track down the source right now, so I though I'd make an issue and see if anyone else had time or interest to track it down.

sloretz, 2019-05-02:
I'm unable to reproduce the segfault using the script in the issue body after multiple attempts. The original traceback goes to `rclpy_destroy_entity`, but that was removed in #319. OK to close this?

Occasionally I see the `KeyboardInterrupt` be ignored with a traceback

```
Exception ignored in: <function WeakSet.__init__.<locals>._remove at 0x7f1a78d9dbf8>
Traceback (most recent call last):
  File "/usr/lib/python3.6/_weakrefset.py", line 38, in _remove
    def _remove(item, selfref=ref(self)):
KeyboardInterrupt
```

But I haven't figured out why that happens. It seems like it might be a cpython 3.6 bug.

wjwwood, 2019-05-02:
Let's close it and reopen if needed. Thanks!

closed 2019-05-02 by wjwwood