ros2/rclpy #186: Implement Time Features in rclpy

tfoote, 2018-04-26:
This ticket is to track rclpy specific time impelementation requirements

 * [ ] implement a Clock on top of rcl implementation
 * [ ] implement a TimeSource
 * [ ] upgrade Timer to use a Clock on top of rcl implementation
 * [ ] Implement a Rate method using the clock on top of rcl implementation

Nickolaim, 2018-05-11:
I can take this item if nobody is looking at it yet.

tfoote, 2018-05-15:
I don't think anyone is actively working on this. If you'd like to start work on the clock, timer and rate that would be feasible. The time source is blocked on the availability of parameters. If you'd like to pick it up I'd suggest creating a separate ticket for one of the smaller parts to sketch out how you're going to approach it to get feedback before you jump in and implement it.

dhood, 2018-07-16:
Here's some additional detail about the steps that would be required.

- Time class wrapping rcl types (`rcl_time_point`, `rcl_duration`) and functions (`rcl_difference_times`, `rcl_get_time_now`). Conversions to/from builtin_interfaces msg type.
- Clock class wrapping rcl's `rcl_clock_t`, exposing get_now; Node has a clock; demos using node.now() if appropriate.
- TimeSource wrapping rcl; implementation of ROS time support: A node is used to create a TimeSource of "ROS time" (subscription to /clock, use_sim_time support by some means), node's clock is managed by the TimeSource.
- Time jump types (`TimeJump`, `JumpThreshold`) and logic moved from rclcpp to rcl, wrapped in python. Jump handlers in python.
- Timer using specific Clock (rclpy Timer currently wraps an rcl timer that rcl_wait uses; would be adapting to changes in rcl's timer to use a specific clock).
- Addition of Rate in a similar manner to what's done for rclcpp.

stonier, 2018-10-30:
Another one for the list, oneshot timers?

dirk-thomas, 2018-10-30:
A oneshot timer should be straight forward to implement on to of regular timers by cancelling the timer in the first callback.

Please consider to contribute a pull request for this feature.

slessans, 2019-06-08:
@dirk-thomas do you know if anyone took a shot at contributing a simple one shot timer? If not I will give it a try

dirk-thomas, 2019-06-08:
> do you know if anyone took a shot at contributing a simple one shot timer?

Not that I am aware of.

> If not I will give it a try

That would be great.

mhpanah, 2019-07-31:
I'm in need of Rate method to use `sim_time`. Has Rate method been implemented on rclpy yet? Thanks.

jacobperron, 2019-08-21:
> I'm in need of Rate method to use sim_time. Has Rate method been implemented on rclpy yet?

Nope. PRs welcome ;)

PR #443 (2019-10-14): Add Rate
Resolves #186

This adds a `Rate` class. It's a tool for running a loop at a constant rate.

Plus:
* Made context shutdown callbacks use `weakref.WeakMethod()`.
  * Should prevent the callback list from growing infinitely.
* Added `Handle.__bool__()`.
  * Used to avoid sleeping if the timer was destroyed.

Comparison with ROS 1:
* ROS 1: `rate = rospy.Rate(10)`; ROS 2 `rate = node.create_rate(10)`.
* Both raise `ROSInterruptException` if the client library has been shutdown
* ROS 1: may raise `ROSTimeMovedBackwardsException`; ROS 2: uses [timer time jump logic](https://github.com/ros2 [... cut]
closed 2019-10-14 by sloretz