ros2/rclpy #189: Clock class

Nickolaim, 2018-05-31:
This is a portion of work from #186.  I propose the first step to be an implementation of Clock class, similar to what's done in [rclcpp/clock.hpp](https://github.com/ros2/rclcpp/blob/eed59992211e65d3d4699a2be092c8d3b992102f/rclcpp/include/rclcpp/clock.hpp) and in [rclcpp/clock.cpp](https://github.com/ros2/rclcpp/blob/eed59992211e65d3d4699a2be092c8d3b992102f/rclcpp/src/rclcpp/clock.cpp) without jumps. 

I found [WallTimer](https://github.com/ros2/rclpy/blob/1ef2924ef8e154c0553edf0fdba4840b08b728f8/rclpy/rclpy/timer.py) class in rclpy, but I could not figure out a relationship between clock and timer.

@tfoote - I am looking forward to your feedback.

tfoote, 2018-05-31:
I'm not sure it would make sense to implement the Clock class without the jump support. Otherwise users won't be able to know that time has skipped when the clock has changed. The class is trivial enough that the jump API and ability to pump it from a time source are important for confirming that the API is set that I'd rather not try to review just the simpler API/structure before the higher level functions are confirmed to work.

The timer should have a clock associated with it to test for how long has elapsed. It's ticketed in rclcpp here: https://github.com/ros2/rclcpp/issues/465

Nickolaim, 2018-06-02:
I see.  I don’t feel comfortable I can have it implemented in reasonable time.  I will pass on this ticket.

What’s the right process? Just close this issue? Your original ticket still has all the tasks.

tfoote, 2018-06-02:
Since we have the other ticket with the tracking info we could close this out. If you do find some time in the future this could be a good place to come back to. You could also start the implementation of the clock, but we just wouldn't plan to merge it until the other parts were implemented.

closed 2018-06-02 by Nickolaim