ros2/rclpy #190: rclpy should provide a method to check if initialized

wjwwood, 2018-06-08:
## Feature request

#### Feature description

`rclpy` should provide a method to indicate if `rclpy.init()` has been called since the program started or since the last time `rclpy.shutdown()` was called.

#### Implementation considerations

This could also be served by provided a specific exception to raise when called too many times, currently it raises a generic `RuntimeError`, but I think a func to tell without raising an exception would also be good to have.

Should work after calling `rclpy.shutdown()` after originally calling `rclpy.init()` the first time.

sloretz, 2018-08-14:
Thanks @hemantku 

I'm not able to assign you to the ticket, but go ahead and start on it anyways. I would be happy to answer questions you may have if you encounter issues.

hemantku, 2018-08-15:
#224

hemantku, 2018-08-24:
@sloretz : Is there an reference architecture doc for rclpy just like launch: https://github.com/ros2/launch/blob/master/launch/doc/source/architecture.rst ? For example, How is the state stored in each node, using the lifecycle node? or is it something else

sloretz, 2018-08-27:
> Is there an reference architecture doc for rclpy just like launch

Not that I know of. There are slides about the architecture leading up to rclpy py starting at slide 30 here: https://roscon.ros.org/2016/presentations/ROSCon%202016%20-%20ROS%202%20Update.pdf

> For example, How is the state stored in each node, using the lifecycle node

What kind of state? There is a pycapsule holding a pointer to an `rcl_node_t` on `node._handle`.

suab321321, 2019-08-17:
I would like help on this on

suab321321, 2019-08-17:
I would like to help on this

pr-db, 2021-08-12:
@wjwwood Hello, I'm new to open source and love working with ros. How can I approach this issue?

sloretz, 2021-11-23:
I think this feature already exists as `rclpy.ok()`

https://github.com/ros2/rclpy/blob/77949d2c54b8104e48a94d0ea83c7049e18a8809/rclpy/rclpy/__init__.py#L55

Implementation

https://github.com/ros2/rclpy/blob/77949d2c54b8104e48a94d0ea83c7049e18a8809/rclpy/rclpy/utilities.py#L49-L52

EricCousineau-TRI, 2023-11-10:
Yeah, I think @sloretz is right - worth closing this issue now?

fujitatomoya, 2023-11-13:
@EricCousineau-TRI it would be appreciated if you could review https://github.com/ros2/rclpy/pull/1198

sloretz, 2023-12-15:
Closing this issue as complete

closed 2023-12-15 by sloretz