ros2/rclpy #213: Memory leak when publishing complicated message

gong0029, 2018-07-31:
When I publish complicated message, the ros message could not be successfully destroyed, which will lead to memory leak!

When I define complicated message type, which includes variable-length array of submessage,  the message could not be successfully destroyed.

You can reproduce it by the following three steps:
(1)Download the code here: https://github.com/gong0029/zoro_test
(2)Compile the three packages: newzoro_msgs, newzorotmp_msgs and pub_and_sub_py
(3)Run the command: ros2 run pub_and_sub_py talkerN1

You will see the memory is keep on increasing.

I think the reason is that:

The message (ZoroMsg) includes variable-length array of submessage (newzorotmp_msgs/TmpMsg[] mdata).
However, in the file "https://github.com/ros2/rclpy/blob/master/rclpy/src/rclpy/_rclpy.c ", Line 1159,  
the function destroy_ros_message(raw_ros_message) could not fully free the field mdata.

dirk-thomas, 2018-07-31:
I assume you are using ROS 2 Ardent or Bouncy (but without the first patch release). Maybe you can clarify that?

The [first patch release of Bouncy](https://discourse.ros.org/t/new-packages-for-ros-2-bouncy-2018-07-20/5421) contains a fix for that problem, see #203. If you update to that version the memory leak should be resolved.

If that is the case please close this ticket or provide additional information if the problem still exists with the latest version.

gong0029, 2018-08-01:
Thanks very much!

sloretz, 2018-08-03:
I ran `ros2 run pub_and_sub_py talkerN1` using Bouncy patch release 1, and it appears the leak is fixed. Closing.

closed 2018-08-03 by sloretz