ros2/rclpy #219: rclpy not recognized when cross compiled for an arm processor

WilliamLNelson, 2018-08-10:
I am currently trying to make ros2 work on a new raspberry pi zero w using arch Linux and a cross compiler. After fixing multiple errors and bugs, there is only one problem left. Trying to run ros2 gives the following error:

Failed to load entry point 'launch': No module named 'rclpy._rclpy'
Failed to load entry point 'echo': No module named 'rclpy._rclpy'
usage: ros2 [-h] Call `ros2 <command> -h` for more detailed usage. ...

We have been recording our process here if anyone wants to try and reproduce the error: https://github.com/austinstig/ros2-raspberry-pi-zero-w

Any help would be greatly appreciated.

Thanks!

dirk-thomas, 2018-08-10:
We kindly ask to raise questions like this on [answers.ros.org](http://answers.ros.org/) instead (see the [ROS support guidelines](http://wiki.ros.org/Support)).

The rational for asking all kind of questions in a single place is:
* There are many more people reading questions there so your chances on getting an answer and in a timely manner are much higher.
* In the future other users will search there for similar problems and can find your question and the potential answers.
* The issue tracker in this repo is used to track bugs, feature requests, etc. But your question is not such an action item.

Therefore I will close this ticket here for now. If you have asked your question on answers.ros.org please feel free to add a link to the question to this ticket so that future readers can find the related question.

closed 2018-08-10 by dirk-thomas

WilliamLNelson, 2018-08-10:
thanks, I am the one who made that post as well. From what I can tell, the issue is closely linked to rclpy and the people who viewed the page on ros answers weren't sure what the problem was. Therefore, I thought I'd ask people here who might be more knowledgeable when it comes to rclpy specifically.