ros2/rclpy #240: Respect the use_sim_time parameter for rclpy nodes

dhood, 2018-09-20:
There is support for Clocks that use ROS Time as described in [this design document](http://design.ros2.org/articles/clock_and_time.html). It was added in [this PR](https://github.com/ros2/rclpy/pull/210).

Clocks can be set to have ROS time "active" (will use the value seen on the /clock topic as its time) by calling `time_source.ros_time_is_active = True`.

A node has a time source associated with it (added in https://github.com/ros2/rclpy/pull/210). At the time of that PR, parameters were not available in rclpy, so the node's time source does not respond to the `use_sim_time` parameter:
https://github.com/ros2/rclpy/blob/62012d3650de790d89c725232451c4e998b396fb/rclpy/rclpy/node.py#L97

As of https://github.com/ros2/rclpy/pull/214, parameters are available in rclpy.

Tasks to get rclpy nodes to respond to the `use_sim_time` parameter:

- [ ] Add a [parameter callback](https://github.com/ros2/rclpy/pull/228) that detects if `use_sim_time` has been set on the node.
- [ ] Add tests that the node's clock output matches what's expected based on the combination of `use_sim_time` parameter and the presence of a `/clock` publisher.

shivangg, 2018-09-25:
@dhood I'd like to work on this issue. Can you assign it to me?

sloretz, 2018-09-25:
@shivangg Thanks!

I can't assign the issue to you (isaacs/github#100), but your contribution is still very welcome. When you make a pull request add the text `connects to ros2/rclpy#240` to the PR description. That will link the PR and issue on the [waffle board](https://waffle.io/ros2/ros2).

mkhansen-intel, 2018-11-14:
@shivangg - I'm in need of this functionality now for testing ROS2 Navigation, did you get this to work?

jacobperron, 2019-04-01:
Resolved by #297

closed 2019-04-01 by jacobperron