ros2/rclpy #241: "AttributeError: 'Time' object has no attribute 'nanoseconds'" when using message_filters

ahuizxc, 2018-09-30:
## Bug report

**Required Info:**

- Operating System:
  - 
- Installation type:
  - 
- Version or commit hash:
  - 
- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue

```
when using message_filters python package, I got AttributeError: 'Time' object has no attribute 'nanoseconds', this caused by my rclcpp version isn't match the message_filter version, so i tried latest-release branch, but it also doesn't have this  module. At last, I tried the master branch which has "nanoseconds", but the compilation was not successful. 
```

#### Expected behavior

#### Actual behavior

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

mikaelarguedas, 2018-09-30:
Thanks @ahuizxc for reporting. Please fill out the issue template with all the requested information and steps to reproduce the issue

ahuizxc, 2018-10-08:
just want to know which branch should i use if i want to use the time module :) thanks~

dirk-thomas, 2018-10-08:
Please include exact information as outlined in the issue template: what you are using (links to repos including the branch name), your platform, what you are doing and a 1-to-1 of the error message / stacktrace you are seeing.

ahuizxc, 2018-10-09:
when i want to use the clock module, i got the error:

    from rclpy.clock import ROSClock
ModuleNotFoundError: No module named 'rclpy.clock'

dirk-thomas, 2018-10-09:
Assuming that you are using the latest release (which you haven't been able to clarify) this is expected since the Python module `rclpy.clock` has been added after the latest release. You will need to use the latest code from the master branch to use this feature.

I will go ahead and close the ticket.

closed 2018-10-09 by dirk-thomas

ahuizxc, 2018-10-09:
when i use the latest code from the master branch, i got this error:

src/ros2/rclpy/rclpy/src/rclpy/_rclpy.c:1210:61: error: incompatible type for argument 4 of ‘rcl_timer_init’
   rcl_ret_t ret = rcl_timer_init(timer, clock, period_nsec, NULL, allocator);

dirk-thomas, 2018-10-09:
> when i use the latest code from the master branch, i got this error:

What version of the dependencies are you trying to use, e.g. `rcl`?