ros2/rclpy #244: On shutdown hooks

stonier, 2018-10-20:
## Feature request

#### Feature description

Something like the `rospy.on_shutdown()` handle. 

#### Implementation considerations

Primary use case here is so that authors of scripts/applications do not need to remember a long sequence of magic shutdown calls. e.g.

```
try:
    rclpy.spin(node)
except KeyboardInterrupt:
     pass
foo.shutdown()
bar.shutdown()
node.destroy_node()
rclpy.shutdown()
```

Instead, the shutdown methods would register themselves in a setup phase inside the class and subsequently the author of the main script need not worry about the shutdown methods at all.

clalancette, 2018-10-25:
@stonier Based on your latest comment, I'm going to close this out for now.  If you feel like this is something that core rclpy needs to provide in the future, please feel free to reopen.

closed 2018-10-25 by clalancette

stonier, 2018-10-26:
I'd be interested to know if there was an explicit reason it was made available in the ROS 1 api. We could perhaps trawl the rospy issues and commit history to see if there is anything there.

clalancette, 2018-10-26:
Unfortunately, it looks like in ros_comm that code comes from this commit: https://github.com/ros/ros_comm/commit/5ac90fafe3 ,which looks like it is probably the port from sourceforge -> github in 2010.  To go further back than that, we'd have to go trawling through the original sourceforge stuff, which I don't think is worth it.  Maybe @tfoote has some recollection on why it was originally added.

chapulina, 2019-07-16:
On `gazebo_ros_pkgs` we have a use case where we want to make a service call right before shutdown. On ROS 1 the `on_shutdown` hook was being used:

https://github.com/ros-simulation/gazebo_ros_pkgs/blob/3cd100e20ba09245be01ad674248aa31a46ad271/gazebo_ros/scripts/spawn_model#L315-L317

If the hook isn't ported, is there an alternative way to achieve this?

That script is being migrated to ROS 2 on https://github.com/ros-simulation/gazebo_ros_pkgs/pull/948/