ros2/rclpy #246: add resolved_name() method to Publisher

dirk-thomas, 2018-11-07:
Requested by @stonier.

crystaldust, 2020-05-20:
Hi, community. Do we have any detailed thoughts or designs on this feature? I would like to help

sloretz, 2020-05-28:
@crystaldust thanks for taking it up!

I think @dirk-thomas is asking for a method to get the name of a publisher *after* all remap rules have been applied. I think that means adding a `resolved_name()` method to [the publisher class](https://github.com/ros2/rclpy/blob/master/rclpy/rclpy/publisher.py) that internally calls the C function [`rcl_publisher_get_topic_name()`](https://github.com/ros2/rcl/blob/0de31d04830f0b33a3adf4a87e179e0d0d76555b/rcl/include/rcl/publisher.h#L431-L456). It will need a method added to the `rclpy` CPython extension. I think `rclpy_get_publisher_logger_Name()` would be a good function that could be studied/copied/adapted for this.

* https://github.com/ros2/rclpy/blob/044ff46603940444432f938df10450bca8d7c4c5/rclpy/src/rclpy/_rclpy.c#L5443-L5446
* https://github.com/ros2/rclpy/blob/044ff46603940444432f938df10450bca8d7c4c5/rclpy/src/rclpy/_rclpy.c#L919-L946

Though, that doesn't show the final pure-python step. For that you can look at `assert_livelines()`: https://github.com/ros2/rclpy/blob/044ff46603940444432f938df10450bca8d7c4c5/rclpy/rclpy/publisher.py#L87-L95

crystaldust, 2020-05-29:
Thanks @sloretz ! I'll look into the code and try, the info is really helpful.

crystaldust, 2020-06-08:
Hi, @sloretz , I've made a basic implementation on the `resolved_name` method to call `rcl_publisher_get_topic_name()`, code [here](https://github.com/crystaldust/rclpy/commits/publisher-resolved-name).

Btw, I made a test to run a node with remapping rules calling `ros2 run PKG_NAME EXECUTABLE_NAME --ros-args -r FROM:=TO` on topics, it also works. 

I see there is a ros2 component design, as I understand, the component is a specific node which is loaded into a container and can receive the remapping rules at runtime, but I'm not able to find some rclpy examples to test it, for now, there is only a [composition example](https://github.com/ros2/demos/tree/master/composition) which is in cpp and calls a `RCLCPP_COMPONENTS_REGISTER_NODE` macro, and I can't find equivelent things on rclpy project. But the container/component mechanism seems to be implemented in rcl, topic names are updated in rcl lib when the remapping rules are being applied, so calling `rcl_publisher_get_topic_name` should always work. Is there some demos for rclpy component that I can work with to make a test?

sloretz, 2020-06-08:
> Hi, @sloretz , I've made a basic implementation on the resolved_name method to call rcl_publisher_get_topic_name(), code here.

Fantastic :tada:

Mind [opening a pull request](https://github.com/ros2/rclpy/compare/master...crystaldust:publisher-resolved-name)? Then I will be able to use Github's UI to give feedback :)

> Is there some demos for rclpy component that I can work with to make a test?

Not that I'm aware of. You're right there is a [design for composition](http://design.ros2.org/articles/roslaunch.html#process-with-multiple-nodes) that was designed to allow composing Python nodes, but I don't think anyone has written a Python composable node container.

crystaldust, 2020-06-09:
I've opened the PR https://github.com/ros2/rclpy/pull/568

sloretz, 2020-06-16:
Fixed by @crystaldust in #568 :tada:

closed 2020-06-16 by sloretz