ros2/rclpy #267: Timer tests fail sporadically

durko, 2019-01-28:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 16.04, 18.04
- Installation type:
  - source
- Version or commit hash:
  - master, 0fafaec6b3764654413d0ce37839d7c7178245af
- DDS implementation:
  - any
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue
- Run the test suite 
- Timer test `test_timer_zero_callbacks100hertz` in https://github.com/ros2/rclpy/blob/0fafaec6b3764654413d0ce37839d7c7178245af/rclpy/test/test_timer.py fails sometimes

In our CI the error happens quite often. On an idle machine it is harder to catch, but even on an otherwise idle 32 core machine that runs the test in a constant loop it will fail occasionally.

#### Expected behavior
Tests succeed always, regardless of testing environment.

#### Actual behavior
Test fails.

#### Additional information
The following considerations go maybe more into questions of the general approach towards unit testing:

- Some tests in `test_timer.py` are run multiple times with different timings, e.g. 10Hz and more aggressive 100Hz.
- As far as a I can see these tests utilize the code being tested in exactly the same manner. For clarity: What I mean by this is, that regardless of the frequency parameter, the code takes exactly the same code path, every branch taken is the same in both scenarios.
- If my impression in the previous point is correct, a test with the relaxed timings of 10Hz (or even more relaxed timings), would already give us all information about the correctness of the code being tested. Additional tests with higher timer frequencies cannot give us any more insights about the correctness of the code, as they are running though exactly the same logical paths.
- However, while the additional tests are not needed for testing *correctness*, they do give us useful insights about *performance* limits.

If the above considerations are correct, than it would make sense to split **unit tests** that ensure **correctness** from **performance tests** that **benchmark** the underlying computational hardware.

Generally unit tests are are most useful when working with the source, performance tests on the other hand are also useful for end users using binaries to figure out if their target hardware platform has the required performance.

I am not sure if this is the place and time to start a discussion like this, so feel free to redirect me here.

In the spirit of the above *for this issue* I would suggest to move towards running the timer tests only once and use relaxed timings, so that even busy CI machines are able to verify correctness of the code/build.

rotu, 2020-03-14:
This is very linked to the CPU load. I have a 8-cpu, 16-core machine running Ubuntu. By stressing 5 cores (using `stress -c 5`, I can get occasional failures in `test_number_callbacks` and `test_cancel_reset`. As that number goes up, these tests fail more and more frequently.

Unless the CI is run with dedicated CPU resources, I don't think we can expect these tests to be reliable as written. I think that also the code is wrong - if a Timer is unable to maintain the requested rate with reasonable jitter in real-life code, there should be user-visible warnings.

ivanpauno, 2020-06-16:
> If the above considerations are correct, than it would make sense to split unit tests that ensure correctness from performance tests that benchmark the underlying computational hardware.

I agree with that, unit tests should only check correctness and not measure performance.
@durko if you want, you can propose deleting the high rate tests.

dirk-thomas, 2020-07-15:
The test in question has been disabled on Windows since it test unrealistic expectation. Based on that I will go ahead and close this ticket.

closed 2020-07-15 by dirk-thomas