ros2/rclpy #279: Executor cannot handle asyncio.sleep() in coroutine

jacobperron, 2019-03-05:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 16.04 and Ubuntu 18.04
- Installation type:
  - source
- Version or commit hash:
  - be4bc88
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue

With a ROS 2 workspace built and sourced:

1. Create `foo_service.py` with the following content:

```python
import asyncio

from example_interfaces.srv import AddTwoInts

import rclpy

async def add_two_ints_callback(request, response):
    response.sum = request.a + request.b

    await asyncio.sleep(1)

    print('Service request handled')

    return response

def main():
    rclpy.init()
    node = rclpy.create_node('foo_service')
    srv = node.create_service(AddTwoInts, 'add_two_ints', add_two_ints_callback)
    rclpy.spin(node)
    rclpy.shutdown()

if __name__ == '__main__':
    main()
```

2. Run the service:

        python3 foo_service.py

3. Run a service client:

        ros2 run examples_rclpy_minimal_client client

#### Expected behavior

Service call succeeds without error.

#### Actual behavior

Service server crashes from assertion error:

```
AssertionError: yield from wasn't used with future
```

Trace:

```
Traceback (most recent call last):
  File "service.py", line 27, in <module>
    main()
  File "service.py", line 22, in main
    rclpy.spin(node)
  File "/home/jacob/ws/actions_ws/install/rclpy/lib/python3.5/site-packages/rclpy/__init__.py", line 119, in spin
    executor.spin_once()
  File "/home/jacob/ws/actions_ws/install/rclpy/lib/python3.5/site-packages/rclpy/executors.py", line 573, in spin_once
    raise handler.exception()
  File "/home/jacob/ws/actions_ws/install/rclpy/lib/python3.5/site-packages/rclpy/task.py", line 207, in __call__
    self._handler.send(None)
  File "/home/jacob/ws/actions_ws/install/rclpy/lib/python3.5/site-packages/rclpy/executors.py", line 324, in handler
    await call_coroutine(entity, arg)
  File "/home/jacob/ws/actions_ws/install/rclpy/lib/python3.5/site-packages/rclpy/executors.py", line 282, in _execute_service
    response = await await_or_execute(srv.callback, request, srv.srv_type.Response())
  File "/home/jacob/ws/actions_ws/install/rclpy/lib/python3.5/site-packages/rclpy/executors.py", line 89, in await_or_execute
    return await callback(*args)
  File "service.py", line 11, in add_two_ints_callback
    await asyncio.sleep(1)
  File "/usr/lib/python3.5/asyncio/tasks.py", line 516, in sleep
    return (yield from future)
  File "/usr/lib/python3.5/asyncio/futures.py", line 362, in __iter__
    assert self.done(), "yield from wasn't used with future"
AssertionError: yield from wasn't used with future
```

#### Additional information

The error appears related to the call to `asyncio.sleep(1)` in the service server callback.
If we replace it with `time.sleep(1)`, the code executes without error.

jacobperron, 2019-03-14:
I think the only way [asyncio.sleep()](https://github.com/python/cpython/blob/23fc0416454c4ad5b9b23d520fbe6d89be3efc24/Lib/asyncio/tasks.py#L469) can be used is if we refactor executors so that they use [asyncio event loops](https://docs.python.org/3.6/library/asyncio-eventloop.html#run-an-event-loop). II'm not well-versed with the decision to define our own `Future` and `Task` classes, but perhaps it's possible to replace them with equivalents in [asyncio](https://docs.python.org/3.6/library/asyncio-eventloop.html#futures).

sloretz, 2019-03-14:
> I'm not well-versed with the decision to define our own Future and Task classes, but perhaps it's 
possible to replace them with equivalents in asyncio.

There are two standard modules for executors/tasks/futures: [concurrent.futures](https://docs.python.org/3/library/concurrent.futures.html) which is thread safe but does not support coroutines, and [asyncio](https://docs.python.org/3/library/asyncio.html#module-asyncio) which supports coroutines but [is not threadsafe](https://docs.python.org/3/library/asyncio-dev.html#concurrency-and-multithreading). The rclpy executor supports both multiple threads and coroutines, so it has to have its own :(

TheDelus, 2021-05-21:
Are there plans to support pythons [async and await syntax](https://www.python.org/dev/peps/pep-0492/) in the future? Not beeing able to use pythons own system of coroutines and async execution is one of a the big things holding us back migrating to ROS2. With ROS1 we use the workaround provided by [locusrobotics/aiorospy](https://github.com/locusrobotics/aiorospy).

Sorry for hijacking this is issue, but it appears to be the most fitting place for my question.

sloretz, 2021-05-26:
@TheDelus `async` coroutines with `await` can be used in rclpy now for pretty much any callback (subscription callback, timer, guard condition, etc). I don't know of any demos doing that, [but it is tested in rclpy](https://github.com/ros2/rclpy/blob/b72a05bd3fb3ac7d27a8da359571afbbfec07f19/rclpy/test/test_executor.py#L154). There's a hidden requirement that if all coroutines are `await`ing something, they'll only be woken again when something wakes the wait set. Meaning, any coroutines you make should be waiting for something like a timer or a guard condition or the receipt of a message.

As for this issue, I think I'll close it. `asyncio` forces everything being done in a single thread. Given the desire to let users have rclpy call callbacks in multiple threads, I don't think it can ever be made to fit.

closed 2021-05-26 by sloretz

gstorer, 2021-07-04:
@sloretz I am curious exactly what is missing from asyncio to prevent it being used. Between [Executing code in thread or process pools](https://docs.python.org/3/library/asyncio-eventloop.html#executing-code-in-thread-or-process-pools), [Running in Threads](https://docs.python.org/3/library/asyncio-task.html#running-in-threads) and [Scheduling From Other Threads](https://docs.python.org/3/library/asyncio-task.html#scheduling-from-other-threads) it seems to me like asyncio has all the features required to do what the ROS2 executor is doing.

We have some ROS systems and are looking to build some data collection and management infrastructure to tie them together. We are planning to use services written in modern Python to do this and it would have been nice to have them interface directly with ROS. But not having proper asyncio support might mean we end up having shims to another communications stack instead.

sloretz, 2021-07-06:
@gstorer I haven't seen `to_thread()` before. Maybe it's worth looking at asyncio in Python 3.9 to see if ROS 2 can make more use of it now. How do you envision it being used?

Maybe a dedicated thread would block on `rcl_wait()` and feed tasks to an asyncio event loop? Then anyone who wants to run stuff in another thread would use `to_thread()` and await the future? If you have an idea please open a new issue for it.