ros2/rclpy #290: NoneType in __del__

reed-lau, 2019-03-21:
## Bug report

**Required Info:**

- Operating System:
  Ubuntu 16.04
- Installation type:
  from source
- Version or commit hash:
 bouncy
- DDS implementation:
  fastrtps, opensplice
- Client library (if applicable):
  rclpy

#### Steps to reproduce issue
1. run the following python script
2. Ctrl-C it

```py
import os
import signal

import numpy as np
from sensor_msgs.msg import Image
from threading import Condition
import threading
import copy

import sys
import time

import rclpy
from rclpy.node import Node

from std_msgs.msg import String

def handler_run(data_stream, handler):
    while rclpy.ok():
        handler(data_stream.fetch_lastest())

def show_msg_len(msg):
    if msg:
        print("show_msg_len:",len(msg.data))

class Listener(Node):

    def __init__(self):
        super().__init__('listener')
        self.sub = self.create_subscription(String, 'chatter', self.on_msg)
        self.lastest_msg = None
        self.__new_data_condition = Condition()
    def fetch_lastest(self):
        data = None
        with self.__new_data_condition:
            if self.lastest_msg is None:
                self.__new_data_condition.wait(1)
            else:
                data = copy.copy(self.lastest_msg)
                self.lastest_msg = None

        return data

    def on_msg(self, msg):
        self.get_logger().info('I heard: [%s]' % len(msg.data))
        with self.__new_data_condition:
            self.lastest_msg = msg
            self.__new_data_condition.notify_all()

def main(args=None):
    if args is None:
        args = sys.argv

    rclpy.init(args=args)

    node = Listener()

    t = threading.Thread(target=handler_run,args=(node,show_msg_len))
    t.daemon = True
    t.start()

    rclpy.spin(node)

    node.destroy_node()
    rclpy.shutdown()

if __name__ == '__main__':
    main()
```
#### Expected behavior
```
Traceback (most recent call last):
  File "listener_thread.py", line 79, in <module>
    main()
  File "listener_thread.py", line 71, in main
    rclpy.spin(node)
  File "/home/tusimple/zoro-linux/lib/python3.5/site-packages/rclpy/__init__.py", line 108, in spin
    executor.spin_once()
  File "/home/tusimple/zoro-linux/lib/python3.5/site-packages/rclpy/executors.py", line 530, in spin_once
    handler, entity, node = self.wait_for_ready_callbacks(timeout_sec=timeout_sec)
  File "/home/tusimple/zoro-linux/lib/python3.5/site-packages/rclpy/executors.py", line 516, in wait_for_ready_callbacks
    return next(self._cb_iter)
  File "/home/tusimple/zoro-linux/lib/python3.5/site-packages/rclpy/executors.py", line 422, in _wait_for_ready_callbacks
    _rclpy.rclpy_wait(wait_set, timeout_nsec)
KeyboardInterrupt
```
#### Actual behavior
```
Traceback (most recent call last):
  File "listener_thread.py", line 79, in <module>
    main()
  File "listener_thread.py", line 71, in main
    rclpy.spin(node)
  File "/home/tusimple/zoro-linux/lib/python3.5/site-packages/rclpy/__init__.py", line 108, in spin
    executor.spin_once()
  File "/home/tusimple/zoro-linux/lib/python3.5/site-packages/rclpy/executors.py", line 530, in spin_once
    handler, entity, node = self.wait_for_ready_callbacks(timeout_sec=timeout_sec)
  File "/home/tusimple/zoro-linux/lib/python3.5/site-packages/rclpy/executors.py", line 516, in wait_for_ready_callbacks
    return next(self._cb_iter)
  File "/home/tusimple/zoro-linux/lib/python3.5/site-packages/rclpy/executors.py", line 422, in _wait_for_ready_callbacks
    _rclpy.rclpy_wait(wait_set, timeout_nsec)
KeyboardInterrupt
Exception ignored in: <bound method Executor.__del__ of <rclpy.executors.SingleThreadedExecutor object at 0x7feca2d8cda0>>
Traceback (most recent call last):
  File "/home/tusimple/zoro-linux/lib/python3.5/site-packages/rclpy/executors.py", line 173, in __del__
TypeError: 'NoneType' object is not callable
```
#### Implementation considerations
sometimes, when threading is used, the deconstruct order is not well defined, so in the `__del__`,   https://github.com/ros2/rclpy/blob/28a77736f1d5ffd13a1dc065e066fab3a89f27f2/rclpy/rclpy/executors.py#L205
the imported `_rclpy` is destroyed.
in this case, we wrap the __del__'s content in the `try ... except`, which could avoid the noisy info.

sloretz, 2019-04-24:
Wrapping just the `_rclpy.rclpy_destroy_entity` call (rather than all of `__del__`) in `try`/`except` makes sense to me. Would you be willing to open a PR targeting `master`?

Relevant https://docs.python.org/3/reference/datamodel.html#object.__del__
```
__del__() can be executed during interpreter shutdown. As a consequence, the global variables it needs to access (including other modules) may already have been deleted or set to None.
```

I don't know when the next bouncy patch release will happen, but if it's fixed on `master` I can add it to the list of tickets to backport ros2/ros2#575.

reed-lau, 2019-04-24:
ok I’ll fix it soon.

reed-lau, 2019-05-09:
it was solved in master.

closed 2019-05-09 by reed-lau