ros2/rclpy #341: Follow-up on parameter API

jubeira, 2019-05-07:
This issue is to track some pending comments in #325, which should not modify existing method signatures:

- [ ] Use `Optional` for `declare_parameters` so as not to require `None` values in the tuples ([comment](https://github.com/ros2/rclpy/pull/325#discussion_r280645899)).
- [ ] Improve test cases; add tests for the remaining methods added.
- [ ] Check if name validation for parameters is appropriate.
- [ ] Check for read-only parameters ([comment](https://github.com/ros2/rclpy/pull/325#discussion_r280239402)).
- [ ] Introduce `set_parameter_descriptor` ([comment](https://github.com/ros2/rclpy/pull/325#discussion_r280534305)).
- [ ] Optional signature using `Parameter` as input for `declare_parameter` ([comment](https://github.com/ros2/rclpy/pull/325#discussion_r280534305)).

jubeira, 2019-05-09:
@wjwwood I have some questions about the behavior we are aiming to implement:

- Regarding `set_parameter_descriptor`: would that method be able to modify the descriptor of a `read_only` parameter? 
Right now, a `read_only` parameter can't be undeclared, so there's no way to change its descriptor by re-declaring it. On the other hand, `set_parameter_descriptor` would affect the just the descriptor in the first place, but eventually could affect the parameter as well somehow once ranges are implemented (e.g. suppose the parameter is out of range when the descriptor is set).
- Regarding name validation, is there any actual difference between validating a topic / service name with respect to a parameter name? I don't see a different method at `rcl` level, and the validation in `rclcpp` is [still pending](https://github.com/ros2/rclcpp/blob/0.7.2/rclcpp/src/rclcpp/node_interfaces/node_parameters.cpp#L268). 
Perhaps the current [implementation](https://github.com/ros2/rclpy/blob/0.7.1/rclpy/rclpy/validate_parameter_name.py#L33) is fine; maybe the naming is not 100% accurate? Do you suggest taking any other action on this end?

wjwwood, 2019-05-09:
> Regarding set_parameter_descriptor: would that method be able to modify the descriptor of a read_only parameter?

I'd say, do not allow it.

If we block undeclare (which was intentional, it was meant to be a strong constraint) then we ought to block changing the descriptor too.

I had thought about having a more nuanced setting for `read_only`, like two settings `indestructible`/`destructible` with a three-way option like `immutable`/`read_only`/`mutable`.
Where something `indestructible` could not be undeclared, and something `read_only` could not be set, but could have its description changed (including the mutability setting), but something `immutable` would be fixed in value and type/description.

Obviously that's a bit half-baked, but I avoided it as it was even more complicated, and I was behind on the pr.

> ... eventually could affect the parameter as well somehow once ranges are implemented (e.g. suppose the parameter is out of range when the descriptor is set).

This is part of the reason I didn't implement it in the first pr. I didn't know how to handle the case where the value met the previous constraints but didn't match the new ones.

I had thought, maybe I could just blindly update the descriptor, then redo the set of the value, so that it would reevaluate the value after updating the descriptor. However, if that failed, then you'd be left with the original value and the new descriptor, which is what you were trying to avoid in the first place.

By forcing the user to undeclare the parameter first, you don't have this problem because you're starting from a valid state (parameter doesn't exist) before you try to set the descriptor or value.

I'm not sure how to handle the corner cases here... Might be worth asking others to chime in.

> Regarding name validation

I avoided this because we have never addressed this topic in a design document, and still need to do so. Right now parameter names are all relative to the node and there's no way to specify a parameter in a globally unique way, i.e. including the node on which it lives.

My personal opinion is that it should look like this:

```
rosparam://node_ns.node_name/some_parameter_ns/controller1/pid_values/p
```

So that it is _host-oriented_ rather than _content-oriented_ like the topics and services, and where the host is a node, who's name and namespace are separated by `.` rather than `/`.

Right now, it's node names and namespaces which are separated by `/` and parameters which are separated by `.`.

> is there any actual difference between validating a topic / service name with respect to a parameter name?

The separator is different. A topic name is like `/foo/bar` and a parameter name is like `foo.bar`.

jubeira, 2019-05-10:
Thanks for sharing those thoughts!

> I'd say, do not allow it.

:ok_hand: I didn't have a strong opinion, but this seemed the most reasonable thing to do to me as well.

> This is part of the reason I didn't implement it in the first pr. I didn't know how to handle the case where the value met the previous constraints but didn't match the new ones.

These could be two options for that (some more opinions are welcome as you say):
- Rejecting the new descriptor if the current value doesn't comply with it (e.g. return a `bool` with the result).
- Provide an optional default together with the descriptor, which shall be used in the case the current value doesn't comply with the new descriptor; maybe return the value that remains. Then:
  - If the existing value complies with the new descriptor, leave it as it is and set the new descriptor. Return the existing value.
  - If the existing value doesn't comply with the new descriptor but the provided default does, set the value to the default, set the descriptor to the new one and return the default value.
  - If neither the existing value nor the provided default comply with the new descriptor, leave the old value and descriptor and raise an exception. 

The first option is simpler, but the second one is a bit more versatile IMO. I'd be slightly inclined for the second one.

> I avoided this because we have never addressed this topic in a design document, and still need to do so. Right now parameter names are all relative to the node and there's no way to specify a parameter in a globally unique way, i.e. including the node on which it lives.

Sounds like this should be something to be solved at `rcl` level and then used in `rclcpp` and `rclpy` like the current topic name validation. Shall we change [this call](https://github.com/ros2/rclpy/blob/0.7.1/rclpy/rclpy/validate_topic_name.py#L36) for a "non-empty" check like in `rclcpp`, or would you suggest something else for the feature freeze?

PR #345 (2019-05-15): Parameter handling improvements.
Closes #341.

There are some details that still need a bit of attention (see https://github.com/ros2/rclpy/issues/341#issuecomment-491292717). The PR includes a proposal for `set_descriptor` as described in the comment.

On the other hand, I'd like some input about the optional signature using `Parameter` as input for `declare_parameter` (see [comment](https://github.com/ros2/rclpy/pull/325#discussion_r280534305)). Is it worth implementing it even if it only applies to `declare_parameter` but not `declare_parameters`?

Other than that, this should be ready for  [... cut]
closed 2019-05-15 by wjwwood