ros2/rclpy #360: Nodes ignore Ctrl-C when using Opensplice on Windows

hidmic, 2019-05-24:
## Bug report

**Required Info:**

- Operating System:
  - Windows (Server) 10
- Installation type:
  - Binary fat archive
- Version or commit hash:
  - [Dashing pre-release](https://ci.ros2.org/view/packaging/job/packaging_windows/1382/) 
- DDS implementation:
  - ADLink Opensplice
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue

Run any Python node

```
ros2 run demo_nodes_py talker
```

Then, attempt to Ctrl-C it.

#### Expected behavior
Process acknowledges that it received a signal and terminates gracefully

#### Actual behavior
Process ignores the signal entirely and can't be killed.

ivanpauno, 2019-05-28:
I have reproduced it too

seanyen, 2019-05-28:
Can it be hanging at somewhere when the process is trying to exit? Just as a reference, I tried the same Dashing binaries plus the installation from this [page](https://index.ros.org/doc/ros2/Installation/Crystal/Windows-Install-Binary/). I can terminate the command by ctrl+c:

```console
C:\Windows\System32>ros2 run demo_nodes_py talker
[INFO] [talker]: Publishing: "Hello World: 0"
[INFO] [talker]: Publishing: "Hello World: 1"
[INFO] [talker]: Publishing: "Hello World: 2"
[INFO] [talker]: Publishing: "Hello World: 3"
[INFO] [talker]: Publishing: "Hello World: 4"
[INFO] [talker]: Publishing: "Hello World: 5"
[INFO] [talker]: Publishing: "Hello World: 6"
Traceback (most recent call last):
  File "c:\dev\ros2\lib\demo_nodes_py\talker-script.py", line 11, in <module>
    load_entry_point('demo-nodes-py==0.7.1', 'console_scripts', 'talker')()
  File "c:\dev\ros2\Lib\site-packages\demo_nodes_py\topics\talker.py", line 43, in main
    rclpy.spin(node)
  File "c:\dev\ros2\Lib\site-packages\rclpy\__init__.py", line 185, in spin
    executor.spin_once()
  File "c:\dev\ros2\Lib\site-packages\rclpy\executors.py", line 663, in spin_once
    handler, entity, node = self.wait_for_ready_callbacks(timeout_sec=timeout_sec)
  File "c:\dev\ros2\Lib\site-packages\rclpy\executors.py", line 649, in wait_for_ready_callbacks
    return next(self._cb_iter)
  File "c:\dev\ros2\Lib\site-packages\rclpy\executors.py", line 549, in _wait_for_ready_callbacks
    _rclpy.rclpy_wait(wait_set, timeout_nsec)
KeyboardInterrupt
```

nuclearsandwich, 2019-05-28:
@seanyen thanks for taking a look! From talking with @hidmic my recollection is that attempting to attach Visual Studio crashed Visual Studio itself :scream_cat:

What's your Windows 10 version and which version of OpenSplice did you try with? Dashing is using the [6.9.190403 version](https://github.com/ADLINK-IST/opensplice/releases/tag/OSPL_V6_9_190403OSS_RELEASE).

seanyen, 2019-05-28:
@nuclearsandwich It's too bad that the debugger even crashed on itself. 😢 Another way is to use [task manager](https://blogs.msdn.microsoft.com/benjaminperkins/2015/09/14/create-a-w3wp-memory-dump-quick-and-easy/) to get a dump file for the process, and then open it by debugger (or upload it here so I might be able to take a look). 

Btw, I updated my environment information. It seems I am using the correct version of OpenSplice.

hidmic, 2019-05-29:
Alright, first of all, thanks for taking a look @seanyen. I can readily say that our Windows version do not match, the VM I'm using runs Windows Server 2019 Datacenter (1809 64 bit). Same Opensplice version but building with Visual Studio 2019. I know @ivanpauno has the same environment, but not sure about @mjcarroll (could you post that info too @mjcarroll?)

We see hanging processes for release builds and heap corruption for debug builds, which may as well be due to ABI mismatches only (as Opensplice does not release debug binaries). I managed to get dump files for both, see [here](https://drive.google.com/file/d/1R3fZ-AnAYRb8MwsZM8-kE-_J3qv57xSL/view?usp=sharing) and [here](https://drive.google.com/file/d/1uUT88K31YDb3ODnnmUafVrjyx_EsSg-l/view?usp=sharing) respectively, but I couldn't keep digging because it looks like we're not packaging symbol files in our debug fat archives :(

mjcarroll, 2019-05-29:
I was able to reproduce the hanging behavior with a source build from `master`

VM Config: 
* Windows 10 Pro x64 (1809)
* Visual Studio Community 2019 (16.1.1)
* https://github.com/ADLINK-IST/opensplice/releases/tag/OSPL_V6_9_190403OSS_RELEASE

seanyen, 2019-05-30:
@hidmic I checked the dump files (for the release builds) and it seems the `Python.exe` is still printing out logs like `[INFO] [talker]: Publishing: "Hello World: 14"` by inspecting the top call frames, so I will assume the program is not really hanging, but somehow just ignore `ctrl-c`, like what you said in the original bug report.

Just wondering what's the terminal\shell as you are running `ros2 run demo_nodes_py talker`? One thing worth checking is if [`ENABLE_PROCESSED_INPUT`](https://docs.microsoft.com/en-us/windows/console/ctrl-c-and-ctrl-break-signals) is disabled in Python by any modules indirectly or it is inherited from the calling shell. That's one way I can think of to reproduce this symptom.

hidmic, 2019-05-30:
> so I will assume the program is not really hanging, but somehow just ignore ctrl-c, like what you said in the original bug report.

Correct, I meant hanging in the sense that it won't acknowledge any Ctrl-C. Sorry for the imprecise statement.

> Just wondering what's the terminal\shell as you are running `ros2 run demo_nodes_py talker`?

Just regular Command Prompt.

> One thing worth checking is if ENABLE_PROCESSED_INPUT is disabled in Python by any modules indirectly or it is inherited from the calling shell.

Hmm, I wonder where that could be. We are using the same sources and binary libraries, but only those running the 1809 64bit variant (us) can reproduce it with one RMW_IMPLEMENTATION (i.e. Opensplice) but not others, and always running on the same terminal (at least in my case). This looks like it can only be tracked down by building everything for debug and monitoring `SetConsoleCtrlHandler` calls. Do you see a way around that @seanyen ?

seanyen, 2019-05-30:
> Just regular Command Prompt.

Can you share more how did you open the command prompt? After some more digging, I found if I am using a command prompt open from Visual Studio Code (by `ctrl-shift-c`), then `ctrl-c` won't work well (see https://github.com/microsoft/vscode/issues/9347). And does the problem manifest in `ros2 run demo_nodes_cpp talker` as well? @hidmic

ivanpauno, 2019-05-31:
> Can you share more how did you open the command prompt? After some more digging, I found if I am using a command prompt open from Visual Studio Code (by `ctrl-shift-c`), then `ctrl-c` won't work well (see [microsoft/vscode#9347](https://github.com/microsoft/vscode/issues/9347)).

In my case, I didn't open a terminal from Visual Studio Code. I opened the usual command prompt from the windows menu bar (pressed the search symbol+typed `cmd`+pressed enter).
I experienced exactly the same error that @hidmic had.

> And does the problem manifest in `ros2 run demo_nodes_cpp talker` as well? @hidmic

The problem is not manifested in `demo_nodes_cpp` `talker`. It only happens with python nodes. e.g.:
It happens too with `ros2 run action_tutorials fibonacci_action_server`.

hidmic, 2019-06-04:
@seanyen @ivanpauno depiction is correct. 

> This looks like it can only be tracked down by building everything for debug and monitoring `SetConsoleCtrlHandler` calls.

I'll assume this is the case and start with it.

hidmic, 2019-06-06:
So far, I've failed to build Opensplice for debug on Windows. I filed an issue upstream:  https://github.com/ADLINK-IST/opensplice/issues/96.

cwyark, 2019-06-13:
I think it's caused by the executor doesn't catch the SIGINT signal and notify rclpy's context to release the resources.

hidmic, 2019-06-13:
Hmm, good point @cwyark, that is indeed a discrepancy in signal handling between `rclcpp` and `rclpy`. Executors are (or at least should) be awaken after guard conditions are triggered, but no context is trashed in the process, nor has ever been in `rclpy`. 

@wjwwood if this is ends up being the root cause, I wonder if it's simply an equivalent of https://github.com/ros2/rclcpp/issues/604.

wjwwood, 2019-06-13:
> @wjwwood if this is ends up being the root cause, I wonder if it's simply an equivalent of ros2/rclcpp#604.

But that issue only occurred if you run sigint twice in quick succession (reentering the signal handler). Is that the case here?

I would believe that there's something incomplete in the signal handling in rclpy, but I don't know if it's the same issue or not.

hidmic, 2019-06-13:
> Is that the case here?

I wouldn't think so initially, but what I meant by equivalent I meant it in the sense of unsafe signal handling code. Just for starters, `rclpy` SIGINT signal handler only does guard condition triggering, which we moved to a background thread in `rclcpp`, and that's only a subset of all the things occurring during signal handling in `rclcpp` (like context shutdown).

> I would believe that there's something incomplete in the signal handling in rclpy,

Basically.

wjwwood, 2019-06-13:
To complicated it further though, I think the Python signal handler is protected in this fashion already, though we do have something in the C extension for rclpy which might be at issue.

MarcelJordense, 2019-08-19:
In opensplice version OSPL_V6_9_190705OSS_RELEASE the problem with the Ctrl-C handling on Windows should have been solved. 
@hidmic could you please verify this.

hidmic, 2019-08-22:
Thanks for the heads up @MarcelJordense. Haven't had the time to circle back on this, will try to do so either tomorrow or Monday.

hidmic, 2019-09-06:
@MarcelJordense apologies for the delay. I can confirm that the Ctrl-C issue is gone in OSPL_V6_9_190705OSS_RELEASE. Thanks! I'll close this one out.

closed 2019-09-06 by hidmic