ros2/rclpy #373: unable to namespace parameters

Karsten1987, 2019-06-19:
## Bug report

Just like in the cpp demos, I'd like to be able to namespace parameters in the form of `<namespace>.<parameter_name>`. However, in python I believe the wrong validation check is performed. Looking at the error output, it looks like the validation for topic names is applied rather than parameters.

#### Steps to reproduce issue

node.declare_parameter(node.get_name() + '.diagnostics_update', 1).value

#### Expected behavior

Able to declare a parameter with namespace in the form of 'namespace.parameter_name'.

#### Actual behavior
```
result = _rclpy.rclpy_get_validation_error_for_topic_name(name)
if result is None:
  return True
error_msg, invalid_index = result
raise InvalidParameterException(name, error_msg, invalid_index)
rclpy.exceptions.InvalidParameterException: ('Invalid parameter name', 
  'test_node.diagnostics_update', "topic name must not contain characters other than alphanumerics, '_', '~', '{', or '}'", 9)
```

wjwwood, 2019-06-20:
@jubeira FYI

jubeira, 2019-06-21:
Thanks for the report.
In `rclcpp` the only validation being performed ATM is checking that the name is not empty:

https://github.com/ros2/rclcpp/blob/e7c463dc74849aa64cb8a74e13f2a57b4ed59862/rclcpp/src/rclcpp/node_interfaces/node_parameters.cpp#L376-L379

Unlike topic name validation, AFAIK there is no function to validate a parameter name at `rcl` level and perhaps it would be a good idea to have it.

I see two ways of proceeding:
- Make the validation just an empty check temporarily as in `rclcpp`, and file a feature request in `rcl` to add topic name validation. Once implemented, start using that functionality in `rclpy`.
  - Note: I suggest making the check empty temporarily rather than waiting for the fix at `rcl` level since this issue might be blocking other developments in the meantime.
- Implement the check in Python. Perhaps this would be faster than implementing it in `rcl`, but it would be not related with the eventual solution in `rclcpp`, which is not consistent with the current solution to validate topic names.

Personally, I suggest going for the first one. @wjwwood thoughts?
/cc @sloretz

joncppl, 2019-06-21:
Not sure if this is this is the same issue or not, but my team has observed what we believe to be incorrect namespacing behaviour in `declare_parameters`.

The parameter keys returned are namespace concatanated with the parameter name, without a separating `.`

ie. The key returned  from `declare_parameters(ns, [(name, default)])` is `ns + name` but we expect `ns + '.' name`

jubeira, 2019-06-21:
> The parameter names returned are namespace concatanated with the parameter name, without a separating .

@joncppl That seems to be a separate issue; thanks for pointing it out.
I'll file a ticket and send a PR with a fix.

jubeira, 2019-06-26:
The problem described in this report has been addressed in #375.
I've opened a ticket to track the implementation for parameter name validation in #380 as a feature request.

@Karsten1987 shall we close this issue?

closed 2019-07-31 by Karsten1987