ros2/rclpy #380: Validate parameter names

jubeira, 2019-06-26:
## Feature request

Validate parameter names on declaration.
The current name validation is simply a non-empty check.

#### Feature description
Complete constraints for parameter names TBD.
Name validation should enable to reject parameters when their name.

#### Implementation considerations
[Nodes](https://github.com/ros2/rclpy/blob/master/rclpy/rclpy/validate_node_name.py) and [topics](https://github.com/ros2/rclpy/blob/master/rclpy/rclpy/validate_full_topic_name.py) have name validations, but parameters don't. 
Following the implementation for node names and topics, it would make sense to implement the solution at RCL level and use that functionality in `rclpy` and `rclcpp`. See [topic and service names](https://design.ros2.org/articles/topic_and_service_names.html) design doc for reference.

suab321321, 2019-07-03:
I want to contribute to this one

nuclearsandwich, 2019-08-15:
@suab321321 please feel free to do so and open a pull request if you have any questions or want early feedback.

suab321321, 2019-08-17:
I have started going through code,will update you soon

suab321321, 2019-08-17:
Do I have to implement this in rcl reposiroty like the ones used for rcl/validate_topic_name?

jubeira, 2019-08-19:
> @nuclearsandwich Do I have to implement this in rcl reposiroty like the ones used for rcl/validate_topic_name,and then using the c api I need to implement this in rospy ,am I correct?

Yes, that would be a good approach.

> And also is the rules for validating a node topic is given in design doc but not of parameters.

AFAIK that is correct.
There is an open discussion [here](https://github.com/ros2/ros2/issues/432) about parameter design and parameter naming in particular; so far there are some assumptions but I think it's not a settled discussion. Perhaps it's better to wait until a design document goes out and then work on this.

ivanpauno, 2019-08-20:
> AFAIK that is correct.
> There is an open discussion [here](https://github.com/ros2/ros2/issues/432) about parameter design and parameter naming in particular; so far there are some assumptions but I [...]

I agree, rules for parameter naming aren't clear now. We should wait until we start implementing https://github.com/ros2/design/pull/241/.

suab321321, 2019-08-20:
Okay I will wait for it.

suab321321, 2019-09-05:
hey there

suab321321, 2019-11-12:
@jubeira are parameters rules clear now..should I start

ivanpauno, 2019-11-18:
> @jubeira are parameters rules clear now..should I start

There is not progress in that discussion at the moment.