ros2/rclpy #393: Support multiple `on parameter set` callbacks

ivanpauno, 2019-07-24:
## Feature request

#### Feature description

Similar to https://github.com/ros2/rclcpp/issues/722.

ivanpauno, 2019-07-24:
By the way, I think `set_parameters_callback` is not a good name.
The idea is to support both `on set` and `on change` callbacks.

The firsts are for rejecting or not the result, the later for updating things based on the new params.

P.S.: There is actually an implementation of the `on change` callback, but not local. You have to subscribe to a topic (at least, it's implemented in `rclcpp`, I guess that here too).

suab321321, 2019-08-18:
@ivanpauno  is this a beginner issue?

ivanpauno, 2019-08-20:
> @ivanpauno is this a beginner issue?

Yes. You have two implement two methods:

```python3
def add_on_set_parameters_callback(callback: Callable[[List[Parameter]], SetParametersResult]) -> None:
  ...
def remove_on_set_parameters_callback(callback: Callable[[List[Parameter]], SetParametersResult]) -> None:
  ...
```
They basically should add/remove a callback from a list.
All the callbacks on that list should be called in the same place where `self._parameters_callback` is called now.

suab321321, 2019-08-20:
@ivanpauno will start working today onwards.

suab321321, 2019-09-11:
xxx

suab321321, 2019-09-11:
okay

ivanpauno, 2019-09-11:
> @ivanpauno will start working today onwards.

Sure, I didn't read your reply before.
You can open a PR with a partial implementation if you want to ask anything.

suab321321, 2019-09-15:
hey @ivanpauno  I just open PR #427 could you please have a look at that confirm me if its okay..

suab321321, 2019-09-16:
hey @ivanpauno  sorry for my disturbance..but when I commit some changes in my branch that should automatically be able to update the PR which is related to that my branch..but here it is creating new pull request..
By mistake I also added another new PR trying to understand this..
And also I m unable to reopen my PR.

Sorry for my mistakes

jacobperron, 2020-02-05:
Closed by #457

closed 2020-02-05 by jacobperron