ros2/rclpy #400: Refactor rclpy signal handling

ivanpauno, 2019-07-30:
## Feature request

#### Feature description

The idea is to make it look more similar to what `rclcpp` is doing, and solve https://github.com/ros2/rclpy/issues/360, https://github.com/ros2/rclpy/issues/398.

#### Implementation considerations

`rclpy` is just triggering a raise condition in its handler, and chaining with the `Python` signal handler. In this way, the wait set is awaken, the `Python` signal handler raises `KeywordInterrupt` and then the execution finishes.
`rclcpp` is shutting down context from a separated thread, specially used for this. It uses a synchronization mechanism to awake that thread when a signal is handled.

ivanpauno, 2019-07-30:
@wjwwood I heard that #360 is solved with the new OpenSplice version, can you confirm?
In that case, I think this is not of much value. I still don't see a reason for having a different behavior in both clients, but I prefer not changing something that works.
#398 can be (almost) surely solved without this refactor.

ivanpauno, 2019-07-30:
@hidmic FYI

ivanpauno, 2019-07-31:
Thinking about this again, if we shut down contexts from another thread we probably shouldn't raise a `KeyboardInterrupt` (we should disable the python default signal handler). That's probably unusual (or at least, it isn't idiomatic).
But in the case the new `OpenSplice` version doesn't solve the problem, this refactor will probably do.

wjwwood, 2019-08-02:
> I heard that #360 is solved with the new OpenSplice version, can you confirm?

I'm not sure, I was just tasked with doing the upgrade, but I think the new version was supposed change something w.r.t. ctrl-c on Windows.

ivanpauno, 2019-09-06:
Closing, as I don't see much value of doing this after the OpenSplice problem was solved:
https://github.com/ros2/rclpy/issues/360#issuecomment-528944306

closed 2019-09-06 by ivanpauno