ros2/rclpy #435: Executor spin_once must check wait_for_ready_callbacks returns None

lesire, 2019-09-25:
- Operating System:
  - Ubuntu 18.04
- Installation type:
  - ubuntu packages
- Version or commit hash:
  - Dashing
- Client library (if applicable):
  - rclpy

When developping a node that has a timer and Action clients, and having a MultiThreadExecutor manage the callbacks, I encoutered the following issue when trying to shutdown the executor from the timer callback:
File "/home/lesire/Work/MUST/ros2_ws/build/oara/oara/actor.py", line 40, in run
    self._executor.spin()
  File "/opt/ros/dashing/lib/python3.6/site-packages/rclpy/executors.py", line 273, in spin
    self.spin_once()
  File "/opt/ros/dashing/lib/python3.6/site-packages/rclpy/executors.py", line 695, in spin_once
    handler, entity, node = self.wait_for_ready_callbacks(timeout_sec=timeout_sec)
TypeError: 'NoneType' object is not iterable

sloretz, 2019-10-10:
Thanks for the bug report! Would you mind adding a reproducible example? That would be helpful for testing. I think I see what happened.

Probably what happened is `spin_once()` got called after shutdown. Inside the generator `_wait_for_ready_callbacks()` it saw `self._is_shutdown` and didn't enter the loop.

https://github.com/ros2/rclpy/blob/b5b4e36c362be044f146df434e73328340c8843f/rclpy/rclpy/executors.py#L447

Since it wasn't a timeout, the generator returned `None` instead of yielding work.

https://github.com/ros2/rclpy/blob/b5b4e36c362be044f146df434e73328340c8843f/rclpy/rclpy/executors.py#L635-L639

The function `wait_for_ready_callbacks` would have gotten `StopIteration`.
https://github.com/ros2/rclpy/blob/b5b4e36c362be044f146df434e73328340c8843f/rclpy/rclpy/executors.py#L663 

Since it had a generator before, it would exit, returing `None`.

https://github.com/ros2/rclpy/blob/b5b4e36c362be044f146df434e73328340c8843f/rclpy/rclpy/executors.py#L653

There seem to be two issues. First `_wait_for_ready_callbacks()` should raise `ShutdownException` instead of returning `None`. Second, `wait_for_ready_callbacks()` should make sure it never returns `None`. Either it should yield work, or it should raise an exception. I can't think of a case where it getting `StopIteration` from `_wait_for_ready_callbacks()` is normal behaviour. It should probably raise a `RuntimeError` in that case.

hamaney, 2020-04-09:
any updates on this?

ivanpauno, 2020-06-19:
> - Version or commit hash:
>   - Dashing

Reopening until a fix gets backported in Dashing.

audrow, 2020-08-03:
@ivanpauno Are you sure that we need to backport #574 to Dashing?  The backport won't be trivial, as I noted in https://github.com/ros2/rclpy/pull/574#issuecomment-646345423: the fix breaks tests in Eloquent (#584) and doesn't  merge cleanly into Dashing.  I suspect we will see at least the same test failures as we did in Eloquent after addressing the merge conflicts in Dashing.

ivanpauno, 2020-08-05:
> @ivanpauno Are you sure that we need to backport #574 to Dashing? The backport won't be trivial, as I noted in #574 (comment): the fix breaks tests in Eloquent (#584) and doesn't merge cleanly into Dashing. I suspect we will see at least the same test failures as we did in Eloquent after addressing the merge conflicts in Dashing.

What I mean is that this is still a bug in Dashing (where the bug is being reported).
We can either fix it, add it to known bugs, or mark the issue as help wanted.

But I don't think that marking the ticket as solved is correct. This is still a bug in the distribution it was reported, and backporting the fix applied on master doesn't seem trivial.

ivanpauno, 2020-08-05:
If you don't have time to try doing the backport @audrow, feel free to assign me and I will try doing it or mark the issue as help wanted if I fail to do so.

audrow, 2020-08-05:
Sounds good. I'll assign you. If you mark it as help wanted, perhaps I'll take a look at in the future.

sloretz, 2022-07-16:
Dashing is EOL. Closing.

PR #574 (2020-06-16): Fix executor behavior on shutdown
Fixes #435 and reverts the changes in #563. 

Note that this PR makes `wait_for_ready_callbacks()` return a `Tuple[Task, WaitableEntityType, 'Node']` as it's type annotation specifies.  Before this method could return `None` and this was implicitly handled in the `spin_once` method for the single and multi thread executors.
closed 2022-07-16 by sloretz