ros2/rclpy #436: get_publisher_names_and_types_by_node() doesn't work

xhuan28, 2019-09-29:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu18.04
- Installation type:
  - source
- Version or commit hash:
  - e962ec9429bf01d842b72b968e7d3f97a59950e7
- DDS implementation:
  - FastRTPS
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue

I created a node and would like to get publisher information of other remote nodes in it. I wrote a snippet as follows:
```
  names = node.get_node_names()
  for name  in node_names:
      print(name)
```
All the nodes are listed successfully. e.g.
remote_node1
remote_node2
current_node

However, when I am trying to get the further information about _remote_node1_, it failed.

```
    info = node.get_publisher_names_and_types_by_node("remote_node1", "/")
    print(len(info))
```

#### Expected behavior
It should return the publisher names and types of the remote node
#### Actual behavior
Nothing is obtained.

I am not sure whether I am misuse the API or it is an issue of this function indeed. Is there anyone help reproduce this?

jacobperron, 2019-10-01:
I can reproduce your issue by running the `demo_nodes_cpp` talker

    ros2 run demo_nodes_cpp talker

and executing the following python script:

```python
import rclpy

def main(args=None):
    rclpy.init(args=args)

    node = rclpy.create_node('foo')

    names = node.get_node_names()
    print('names: {}'.format(names))

    info = node.get_publisher_names_and_types_by_node("talker", "/")
    print('info: {}'.format(info))

    node.destroy_node()
    rclpy.shutdown()

if __name__ == '__main__':
    main()
```

Note, if you pass an empty string for the namespace, instead of "/", then it appears to work fine. E.g. change you example:

```python
info = node.get_publisher_names_and_types_by_node("remote_node1", "")
```

I'm not sure what others think, but I think passing the global namespace explicitly should also work, so I'm marking this as a bug.

xhuan28, 2019-10-10:
@jacobperron 
Thanks for help reproduce this issue. Per your suggestion, I tried to get information by changing the script you provided as follows:
```
info = node.get_publisher_names_and_types_by_node("talker", "")
```
It seems that it doesn't work, either.

> names: ['foo', '_ros2cli_daemon_0', 'talker']
> info: []

jacobperron, 2019-10-10:
@xhuan28 You're right. I was testing with the latest source code, but if I use the Dashing release with Fast-RTPS (the default RMW implementaiton) then using an empty string does not work either.

It appears this is an issue with discovery and Fast-RTPS that must have been fixed post-Dashing. If I try with OpenSplice or Connext, then then my script (with the empty string) works as expected, e.g.

    # shell 1
    RMW_IMPLEMENTATION=rmw_connext_cpp ros2 run demo_nodes_cpp talker

    # shell 2
    RMW_IMPLEMENTATION=rmw_connext_cpp python3 repro_436.py

jacobperron, 2019-10-10:
@xhuan28 Actually, it looks like discovery is just slower in Dashing. Try waiting, to give time for discovery to happen:

```python
import time
import rclpy

def main(args=None):
    rclpy.init(args=args)

    node = rclpy.create_node('foo')

    time.sleep(5)
    names = node.get_node_names()
    print('names: {}'.format(names))

    info = node.get_publisher_names_and_types_by_node("talker", "")
    print('info: {}'.format(info))

    node.destroy_node()
    rclpy.shutdown()

if __name__ == '__main__':
    main()
```

This works for me.

xhuan28, 2019-10-17:
Thanks @jacobperron, it works for me, too.

mabelzhang, 2019-10-17:
Sounds great. Can this ticket be closed?

jacobperron, 2019-10-17:
> Sounds great. Can this ticket be closed?

The method works with an empty string, but there's still the question of whether it should also work with a global namespace (`"/"`). I think it should; we can see other functions handling this case explicitly, e.g.

https://github.com/ros2/rclpy/blob/37c07f60e4544640a65333bb095eb7034f143c1e/rclpy/src/rclpy/_rclpy.c#L4662

I can look into seeing if this can be fixed in a similar way.

jacobperron, 2019-10-17:
False alarm. If we wait to allow discovery before making the graph query, then the `"/"` namespace works as expected.

closed 2019-10-17 by jacobperron

xhuan28, 2019-10-18:
@jacobperron , do you mean if the process sleeps for some time before getting the topic name and type, it works even with "/"?

jacobperron, 2019-10-18:
> do you mean if the process sleeps for some time before getting the topic name and type, it works even with "/"?

Yes, it seems to work for me. Can you confirm?

xhuan28, 2019-11-01:
Confirmed, it works. @jacobperron.