ros2/rclpy #481: dashing v0.7.9 rclpy incompatible with python 3.5

LiaoHongBin, 2019-12-12:
## Bug report

**Required Info:**

- Operating System:
  - Debian 9
- Installation type:
  - from source
- Version or commit hash:
  - dashing (rclpy 0.7.9)
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue
1. Follow the instruction (https://index.ros.org/doc/ros2/Installation/Dashing/Linux-Development-Setup/) to setup ros2_dashing on Debian 9
2. run `ros2 topic list`
#### Expected behavior
expected output:
```
/parameter_events
/rosout
```
#### Actual behavior
actual output:
```
Failed to load entry point 'send_goal': invalid syntax (client.py, line 299)
Failed to load entry point 'list': invalid syntax (client.py, line 299)
Failed to load entry point 'info': invalid syntax (client.py, line 299)
Failed to load entry point 'show': invalid syntax (client.py, line 299)
/parameter_events
/rosout
```
#### Additional information
Actually, it's because the commit #474 #475, which use python 3.6 way to format string, and it's incompatible with python 3.5.

nuclearsandwich, 2019-12-12:
@jacobperron can you please update https://github.com/ros2/rclpy/pull/475 so it is python3.5 compatible?

jacobperron, 2019-12-12:
Yep, on it. I'll have to keep that in mind next time I backport to Dashing.

jacobperron, 2019-12-12:
Should be resolved by #483

closed 2019-12-12 by jacobperron

nuclearsandwich, 2019-12-12:
0.7.10 is now released and is heading to the buildfarm. Thanks for reporting this!

PR #483 (2019-12-12): [dashing] Remove f-strings
They are not compatible with Python 3.5 and accidentally introduced in #475.

Fixes #481