ros2/rclpy #484: Domain ID not changable via ROS_DOMAIN_ID

fujitatomoya, 2019-12-17:
## Feature request

#### Feature description

ROS_DOMAIN_ID environment variable support

#### Implementation considerations

1. ROS_DOMAIN_ID is supported via [rcl](https://github.com/ros2/rcl)
2. rclcpp and rclpy can get domain id included via rcl_node_get_default_options().
3. rclcpp::NodeOptions can override the domain id if necessary.

fujitatomoya, 2019-12-17:
related to https://github.com/ros2/rclcpp/issues/910

ivanpauno, 2019-12-18:
Closing, as I think it has been clarified in https://github.com/ros2/rclcpp/issues/910#issuecomment-566520259.
Please, reopen if not.

fujitatomoya, 2019-12-19:
@ivanpauno 

this is for the same ability with https://github.com/ros2/rclcpp/issues/910 for rclpy. i believe that this is needed for rclpy also not only for rclcpp.

ivanpauno, 2019-12-19:
> @ivanpauno
> 
> this is for the same ability with [ros2/rclcpp#910](https://github.com/ros2/rclcpp/issues/910) for rclpy. i believe that this is needed for rclpy also not only for rclcpp.

Sorry, I didn't understand that.
The title sounds like rclpy is not using `ROS_DOMAIN_ID` environment variable.
Can you update the description and reopen? Thanks!

fujitatomoya, 2019-12-20:
@ivanpauno 

> The title sounds like rclpy is not using ROS_DOMAIN_ID environment variable.

my bad, that confuses you.

> Can you update the description and reopen? Thanks!

will do it right now.

suab321321, 2020-01-02:
@ivanpauno sir I would like to contribute to this.

ivanpauno, 2020-01-02:
For the same reason than here https://github.com/ros2/rclcpp/issues/910#issuecomment-555000410, it's better to wait now. The other set of PRs will be done soon, after that, this will be unblocked.

fujitatomoya, 2020-06-11:
@ivanpauno 

this one too, you can leave it to us.

Barry-Xu-2018, 2020-06-12:
@ivanpauno @fujitatomoya 

After checking current implementation of rclpy, it only supports use default rcl_opitions.   

https://github.com/ros2/rclpy/blob/8f601c69ce9775ee3764c2bcfa2cd22cd9c31a1d/rclpy/src/rclpy/_rclpy.c#L523-L528

If want to support configuring domain ID, I have to expose InitOption to rclpy.  
What do you think ?

fujitatomoya, 2020-06-15:
@Barry-Xu-2018 

i think we do need to expose InitOption to rclpy, i think that's the idea.

Barry-Xu-2018, 2020-06-15:
> @Barry-Xu-2018
> 
> i think we do need to expose InitOption to rclpy, i think that's the idea.

Thank you.

Barry-Xu-2018, 2020-06-16:
@ivanpauno 

Friendly ping.  
If have time, please look at comment.  https://github.com/ros2/rclpy/issues/484#issuecomment-643233546  
I want to hear your opioin.

ivanpauno, 2020-06-29:
> @ivanpauno @fujitatomoya
>
> After checking current implementation of rclpy, it only supports use default rcl_opitions. [...]

Prefer adding an extra keyword argument to the functions with a default:

https://github.com/ros2/rclpy/blob/7116fb4eb51282f706a361ebb25928ee8e0eda3f/rclpy/rclpy/__init__.py#L62
https://github.com/ros2/rclpy/blob/7116fb4eb51282f706a361ebb25928ee8e0eda3f/rclpy/rclpy/context.py#L36

Both should have a new `, domain_id: int = get_default_domain_id()` argument.

In `_rclpy.c`, you will need to write a new `rclpy_get_default_domain_id` function, and extend `rclpy_init` signature with an extra argument.

Barry-Xu-2018, 2020-06-30:
> Prefer adding an extra keyword argument to the functions with a default:
>
> https://github.com/ros2/rclpy/blob/7116fb4eb51282f706a361ebb25928ee8e0eda3f/rclpy/rclpy/__init__.py#L62 [...]

Thanks for your suggestion.

ivanpauno, 2020-06-30:
> Both should have a new `, domain_id: int = get_default_domain_id()` argument.

You could also use the number [here](https://github.com/ros2/rcl/blob/b9cfc243bf018091b5c5bcb5585d19591daf4ae6/rcl/include/rcl/domain_id.h#L29) as a default value, and `rcl` will check the environment variable [here](https://github.com/ros2/rcl/blob/b9cfc243bf018091b5c5bcb5585d19591daf4ae6/rcl/src/rcl/init.c#L147-L154).

Barry-Xu-2018, 2020-07-01:
> You could also use the number [here](https://github.com/ros2/rcl/blob/b9cfc243bf018091b5c5bcb5585d19591daf4ae6/rcl/include/rcl/domain_id.h#L29) as a default value, and `rcl` will check the [...]

Thanks. Got it.

Barry-Xu-2018, 2020-07-02:
@ivanpauno @fujitatomoya 

Please review codes.   
#594

fujitatomoya, 2020-08-06:
CC: @ivanpauno 

i am closing this since https://github.com/ros2/rclpy/pull/596 is merged.

closed 2020-08-06 by fujitatomoya