ros2/rclpy #486: Executor spin_some support

pohzhiee, 2019-12-19:
Similar to rclcpp Executor::spin_some(), which spins the node until all tasks are complete. Is there any easy alternative to achieve this behaviour? Otherwise, when attempting to implement this feature in rclpy, what potential considerations need to be made?

gonzodepedro, 2020-01-16:
I cannot think of an easy alternative to implement that behavior on rclpy. If attempting to implement it I'd take a look at the [discussion around spin_some implementation in rclcpp](https://github.com/ros2/rclcpp/pull/844) and to the [_wait_for_ready_callback](https://github.com/ros2/rclpy/blob/677de6f23d19c232543e822642f1aa30f6ff4bee/rclpy/rclpy/executors.py#L426) method of course.

ivanpauno, 2020-01-30:
@pohzhiee `spin_some` should collect executables once and then execute all of them (without collecting executables again).
A PR adding support to `spin_some` is welcomed.

charlielito, 2021-06-24:
Any update on this? Almost 2 1/2 half year passed :disappointed:

ivanpauno, 2021-07-12:
@charlielito The issue is labeled as `help wanted`.
That means that there isn't someone actively looking to implement it.

PRs are welcomed :smiley:.
I would recommend taking a look at the rclcpp implementation first.
You can ask any question here.

clalancette, 2023-06-05:
> I would like to help with this, but I don't know exactly how to do so. Can you give more information about where to start?

We want a version of `spin_some` in rclpy that acts like the equivalent function in rclcpp.  Thus, the best thing to do here would be to look at the rclcpp implementation of `spin_some`, specifically:

* https://github.com/ros2/rclcpp/blob/f8072f2fa265000a2a1be8f3dbe7e04d9dfb1c91/rclcpp/include/rclcpp/executor.hpp#L285-L298
* https://github.com/ros2/rclcpp/blob/f8072f2fa265000a2a1be8f3dbe7e04d9dfb1c91/rclcpp/include/rclcpp/executor.hpp#L441-L452
* https://github.com/ros2/rclcpp/blob/f8072f2fa265000a2a1be8f3dbe7e04d9dfb1c91/rclcpp/src/rclcpp/executor.cpp#L429-L432
* https://github.com/ros2/rclcpp/blob/f8072f2fa265000a2a1be8f3dbe7e04d9dfb1c91/rclcpp/src/rclcpp/executor.cpp#L442-L478

Note that things are different in Python/rclpy, so the implementation won't look exactly the same, but should provide the same functionality.