ros2/rclpy #491: Failed to load entry point 'test, launch, etc.': No module named 'rclpy._rclpy' on ROS2 Eloquent, macOS Catalina 10.15.2

KKKKKyle, 2020-01-09:
## Bug report

**Required Info:**

- Operating System: macOS Catalina, version 10.15.2
  - 
- Installation type: binaries (following this guide: https://index.ros.org/doc/ros2/Installation/Eloquent/OSX-Install-Binary/)
  - 
- Version or commit hash: ROS 2 Eloquent Elusor
  - 
- DDS implementation: /
  - 
- Client library (if applicable): /
  - 

#### Steps to reproduce issue

After going through all the steps listed in the guide listed above, then run (in bash):
```
tar xf ~/Downloads/ros2-eloquent-20191122-macos-amd64.tar.bz2 
. ~/ros2_eloquent/ros2-osx/setup.bash
ros2 run demo_nodes_cpp talker
```

#### Expected behavior
print out "hello world"

#### Actual behavior
Failed to load entry point 'test': No module named 'lark'
Failed to load entry point 'launch': No module named 'lark'
Failed to load entry point 'info': No module named 'rclpy._rclpy'
Failed to load entry point 'list': No module named 'rclpy._rclpy'
Failed to load entry point 'send_goal': No module named 'rclpy._rclpy'
Failed to load entry point 'show': No module named 'rclpy._rclpy'
Failed to load entry point 'play': No module named 'rclpy._rclpy'
Failed to load entry point 'record': No module named 'rclpy._rclpy'
Failed to load entry point 'list': No module named 'rclpy._rclpy'
Failed to load entry point 'load': No module named 'rclpy._rclpy'
Failed to load entry point 'standalone': No module named 'rclpy._rclpy'
Failed to load entry point 'types': No module named 'rclpy._rclpy'
Failed to load entry point 'unload': No module named 'rclpy._rclpy'
Failed to load entry point 'start': No module named 'rclpy._rclpy'
Failed to load entry point 'status': No module named 'rclpy._rclpy'
Failed to load entry point 'stop': No module named 'rclpy._rclpy'
Failed to load entry point 'get': No module named 'rclpy._rclpy'
Failed to load entry point 'list': No module named 'rclpy._rclpy'
Failed to load entry point 'nodes': No module named 'rclpy._rclpy'
Failed to load entry point 'set': No module named 'rclpy._rclpy'
Failed to load entry point 'info': No module named 'rclpy._rclpy'
Failed to load entry point 'list': No module named 'rclpy._rclpy'
Failed to load entry point 'delete': No module named 'rclpy._rclpy'
Failed to load entry point 'describe': No module named 'rclpy._rclpy'
Failed to load entry point 'dump': No module named 'rclpy._rclpy'
Failed to load entry point 'get': No module named 'rclpy._rclpy'
Failed to load entry point 'list': No module named 'rclpy._rclpy'
Failed to load entry point 'set': No module named 'rclpy._rclpy'
Failed to load entry point 'create': No module named 'catkin_pkg'
Failed to load entry point 'create_key': No module named 'rclpy._rclpy'
Failed to load entry point 'create_keystore': No module named 'rclpy._rclpy'
Failed to load entry point 'create_permission': No module named 'rclpy._rclpy'
Failed to load entry point 'distribute_key': No module named 'rclpy._rclpy'
Failed to load entry point 'generate_artifacts': No module named 'rclpy._rclpy'
Failed to load entry point 'generate_policy': No module named 'rclpy._rclpy'
Failed to load entry point 'list_keys': No module named 'rclpy._rclpy'
Failed to load entry point 'call': No module named 'rclpy._rclpy'
Failed to load entry point 'find': No module named 'rclpy._rclpy'
Failed to load entry point 'list': No module named 'rclpy._rclpy'
Failed to load entry point 'type': No module named 'rclpy._rclpy'
Failed to load entry point 'bw': No module named 'rclpy._rclpy'
Failed to load entry point 'delay': No module named 'rclpy._rclpy'
Failed to load entry point 'echo': No module named 'rclpy._rclpy'
Failed to load entry point 'find': No module named 'rclpy._rclpy'
Failed to load entry point 'hz': No module named 'rclpy._rclpy'
Failed to load entry point 'info': No module named 'rclpy._rclpy'
Failed to load entry point 'list': No module named 'rclpy._rclpy'
Failed to load entry point 'pub': No module named 'rclpy._rclpy'
Failed to load entry point 'type': No module named 'rclpy._rclpy'
[INFO] [talker]: Publishing: 'Hello World: 1'
[INFO] [talker]: Publishing: 'Hello World: 2'
[INFO] [talker]: Publishing: 'Hello World: 3'
[INFO] [talker]: Publishing: 'Hello World: 4'
[INFO] [talker]: Publishing: 'Hello World: 5'
...
#### Additional information
I am pretty new to ros2, but I am guessing even the test works, it still appears to be a bug. When I run 
```
ros2 service
```
here's the output:
```
$ ros2 service
Failed to load entry point 'test': No module named 'lark'
Failed to load entry point 'launch': No module named 'lark'
Failed to load entry point 'info': No module named 'rclpy._rclpy'
Failed to load entry point 'list': No module named 'rclpy._rclpy'
Failed to load entry point 'send_goal': No module named 'rclpy._rclpy'
Failed to load entry point 'show': No module named 'rclpy._rclpy'
Failed to load entry point 'play': No module named 'rclpy._rclpy'
Failed to load entry point 'record': No module named 'rclpy._rclpy'
Failed to load entry point 'list': No module named 'rclpy._rclpy'
Failed to load entry point 'load': No module named 'rclpy._rclpy'
Failed to load entry point 'standalone': No module named 'rclpy._rclpy'
Failed to load entry point 'types': No module named 'rclpy._rclpy'
Failed to load entry point 'unload': No module named 'rclpy._rclpy'
Failed to load entry point 'start': No module named 'rclpy._rclpy'
Failed to load entry point 'status': No module named 'rclpy._rclpy'
Failed to load entry point 'stop': No module named 'rclpy._rclpy'
Failed to load entry point 'get': No module named 'rclpy._rclpy'
Failed to load entry point 'list': No module named 'rclpy._rclpy'
Failed to load entry point 'nodes': No module named 'rclpy._rclpy'
Failed to load entry point 'set': No module named 'rclpy._rclpy'
Failed to load entry point 'info': No module named 'rclpy._rclpy'
Failed to load entry point 'list': No module named 'rclpy._rclpy'
Failed to load entry point 'delete': No module named 'rclpy._rclpy'
Failed to load entry point 'describe': No module named 'rclpy._rclpy'
Failed to load entry point 'dump': No module named 'rclpy._rclpy'
Failed to load entry point 'get': No module named 'rclpy._rclpy'
Failed to load entry point 'list': No module named 'rclpy._rclpy'
Failed to load entry point 'set': No module named 'rclpy._rclpy'
Failed to load entry point 'create': No module named 'catkin_pkg'
Failed to load entry point 'create_key': No module named 'rclpy._rclpy'
Failed to load entry point 'create_keystore': No module named 'rclpy._rclpy'
Failed to load entry point 'create_permission': No module named 'rclpy._rclpy'
Failed to load entry point 'distribute_key': No module named 'rclpy._rclpy'
Failed to load entry point 'generate_artifacts': No module named 'rclpy._rclpy'
Failed to load entry point 'generate_policy': No module named 'rclpy._rclpy'
Failed to load entry point 'list_keys': No module named 'rclpy._rclpy'
Failed to load entry point 'call': No module named 'rclpy._rclpy'
Failed to load entry point 'find': No module named 'rclpy._rclpy'
Failed to load entry point 'list': No module named 'rclpy._rclpy'
Failed to load entry point 'type': No module named 'rclpy._rclpy'
Failed to load entry point 'bw': No module named 'rclpy._rclpy'
Failed to load entry point 'delay': No module named 'rclpy._rclpy'
Failed to load entry point 'echo': No module named 'rclpy._rclpy'
Failed to load entry point 'find': No module named 'rclpy._rclpy'
Failed to load entry point 'hz': No module named 'rclpy._rclpy'
Failed to load entry point 'info': No module named 'rclpy._rclpy'
Failed to load entry point 'list': No module named 'rclpy._rclpy'
Failed to load entry point 'pub': No module named 'rclpy._rclpy'
Failed to load entry point 'type': No module named 'rclpy._rclpy'
usage: ros2 service [-h] [--include-hidden-services]
                    Call `ros2 service <command> -h` for more detailed usage.
                    ...

Various service related sub-commands

optional arguments:
  -h, --help            show this help message and exit
  --include-hidden-services
                        Consider hidden services as well

Commands:

  Call `ros2 service <command> -h` for more detailed usage.
```

jacobperron, 2020-01-17:
macOS Catalina is not a supported platform ([Reference](https://www.ros.org/reps/rep-2000.html#eloquent-elusor-november-2019-november-2020)), therefore I'm inclined to close this issue for now. If you are able to reproduce the problem on Mojave, please comment and we can re-open this ticket.

closed 2020-01-17 by jacobperron

tomeeeS, 2020-04-28:
I just encountered this on Linux 18.4 LTS, following the install tutorial with the apt repository method.