ros2/rclpy #494: test_action_client sometimes times out

jacobperron, 2020-01-09:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04, Windows 10
- Installation type:
  - from source
- Version or commit hash:
  - 9db395545494555aebf3ce67d4b776b27bddd069
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue

Run tests for rclpy.

I've noticed the failure frequently when triggering Windows CI and even a few time running tests locally on Linux. 

Example failure: [image]

ivanpauno, 2020-01-29:
@jacobperron I haven't found a similar error in the last 5 nightly repeated build, do you think this is still a problem?
If that's the case, I can try to configure a windows machine and check if I can reproduce it. If it's just sometimes timing out, we can increase the timeout.

jacobperron, 2020-01-30:
Looks like the last occurrence on the repeated build was [January 10th](https://ci.ros2.org/view/nightly/job/nightly_win_rep/1721/testReport/junit/(root)/rclpy/test_action_client_xunit_missing_result/). I'm not sure what changed since then that would resolve the failure. 

I'm fine closing this and we can re-open if we see another occurrence.

closed 2020-01-30 by jacobperron