ros2/rclpy #496: add publish/take _serialized_message functions to rclpy

Karsten1987, 2020-01-15:
In the style of https://github.com/ros2/rclpy/pull/495 it would be helpful for further (rosbag2) tools to have the capability to publish and receive serialized ros messages.

Suyash458, 2020-01-30:
@Karsten1987 is this still up for grabs? I'd like to work on this

jacobperron, 2020-01-30:
@Suyash458 Great! I'm happy to review any pull requests adding these functions. Let me know if you need any guidance.

Suyash458, 2020-02-10:
Looks like we need an `rclpy_publish_serialized` function in `src/rcply/_rcply.c` which calls `rcl_publish_serialized`. For receiving serialized messages, `rclpy_take_raw` is already implemented in `_rclpy.c`, it just has to be called. I'll submit a PR soon

Suyash458, 2020-02-13:
@jacobperron How can I build rclpy and run the tests?

jacobperron, 2020-02-15:
@Suyash458 To start, you'll need a  version of the latest ROS 2 code by following [these instructions](https://index.ros.org/doc/ros2/Installation/Latest-Development-Setup/). Once you have the source code, you can build and test with colcon ([here's a tutorial](https://index.ros.org/doc/ros2/Tutorials/Colcon-Tutorial/)).

To specifically build rclpy and it's dependencies:

        colcon build --packages-up-to rclpy

And to test rclpy

        colcon test --packages-select rclpy

reed-lau, 2020-02-19:
I thought that:
when we have `publish_raw`, we can introduce `Serialization Tools` at the python application level. For example, we can use `Protobuffer` as the Type-System in our python application. After serializing it, we publish it with `publish_raw`.
What we have to concern is that:
If we just want to use `publish_raw`, when `create_publisher` a `raw` parameter may be needed to avoid the MsgType check. Although, we can make a fake type. Conceptually, when we `publish_raw` no type is needed. a generic `RawType` for `raw` maybe needed.

dirk-thomas, 2020-02-19:
> we can use `Protobuffer` as the Type-System in our python application. After serializing it, we publish it with `publish_raw`.

You might be misunderstanding what the C / C++ functions for raw publishing are doing. You can't provide arbitrary binary data. The underlying RMW implementation requires the passed data to be in the wire representation of the topic. For DDS based implantation that usually means CDR - not protobuf.

reed-lau, 2020-02-19:
@dirk-thomas, thanks so much for your explanation. I get your point.
In my opinion,  sometimes, we can use DDS as the transportation layer without the serialization function/ability.
That means we may need a Publisher/Subscriber semantic without type-aware. (although `std_msgs::String` may be the fake/alternative one).

dirk-thomas, 2020-02-19:
> That means we may need a Publisher/Subscriber semantic without type-aware. (although `std_msgs::String` may be the fake/alternative one).

I think you will need to provide a very detailed rational why you would want to send these data as an opaque blob. With such an approach you loose a lot of the functionality in ROS (unable to introspect the data, loss of semantic, loss of typing of a topic) as well as DDS (unable to apply concepts like keys, content-based filtering, etc.). I almost sounds like an anti pattern to me.

reed-lau, 2020-02-26:
sometimes, we may use protobuffer for its compatibility.
e.g.
1. we record some data in the early days and stored it in a database. the message's definition is something like:
```py
int32 x;
int32 y;
int32 w;
```
2. now we want to re-play it and do some simulation work with a new node. but the message is changed to:
```
int32 x;
int32 y;
```
with the help of potobuffer, we can easily gain compatibility with the help of publishing raw data.

closed 2020-03-23 by jacobperron