ros2/rclpy #501: declare_parameters() should have a possibilty to ignore "ParameterAlreadyDeclaredException"

mganglb, 2020-02-04:
Hello, i am no Expert! So pl forgive me, if i just miss some basic ros2 parameter background knowledge.

i would like to use the method node.declare_parameters([*]). Usually declare one parameter at a time with node.declare_parameter(*).

If i use a *.launch.py file to startup the node, i usually declare the parameters used by the node in the launch file as well. To overcome the problem i just use something like that.

```
try: 
        node.declare_parameter(name, default, descriptor)
        node.get_logger().info("Paramter {} declared to default: {}".format(name, default))
except ParameterAlreadyDeclaredException:
        # this happens e.g when it is declared in launchfile
        node.get_logger().info("Paramter {} already declared: {}".format(name, moya.get_ros_param_value(node, name)))
        pass
```

Since this is not possible, if we declare multiple parameters at once with node.declare_parameters([*]), i would propose an "ignore ParameterAlreadyDeclaredException"- flag

https://github.com/ros2/rclpy/blob/4725106460bc31a0aaa11645847cc4084e7d1e78/rclpy/rclpy/node.py#L441

jacobperron, 2020-02-04:
Could you provide an example of how you are declaring parameters in a launch file? Also, what are you trying to achieve by doing this?

I would avoid calling `rclpy` API directly in your launch files. I suspect you can accomplish what you want with features available in `launch`.

mganglb, 2020-02-05:
i am gonna make more tests on the we and post it. Your example works on my station as well.
I think it is maybe connected to --symlink-install.

clalancette, 2024-12-30:
We never got any additional information here, so I'm going to close this for now.  If you'd still like to pursue it please feel free to reopen.

closed 2024-12-30 by clalancette