ros2/rclpy #507: When I import msg, ide reminds me a error

YoJn, 2020-02-09:
## Code structure

[image]

In my project, I put rclpy and B rcl_interface together, because rclpy import some code from rcl_interface .

## Error

[image]

ide can find that package, but can not find that data structure.

What should I do? Thanks for your help!

YoJn, 2020-02-09:
[image]

tfoote, 2020-02-11:
It looks like you have action_msgs in your source tree. As such the messages will be generated when you run the build. Until then your IDE won't be able to find the definition as the python version has not been generated. So after you run the build once hopefully your IDE will pick up the generated python code. Until then it's likely that you won't be able to resolve the issue. 

Another option would be to have the action_msgs in an underlay workspace that you build from source or install from binaries the default version.

YoJn, 2020-02-11:
> It looks like you have action_msgs in your source tree. As such the messages will be generated when you run the build. Until then your IDE won't be able to find the definition as the python version [...]

Thanks for your help, but I still don't know how to build it.  Does it need me to take the msg files to convert python files? Although I don't know how to convert. 

[abcrclpy.zip](https://github.com/ros2/rclpy/files/4183845/abcrclpy.zip)

tfoote, 2020-02-11:
Yes, you need to have the generators convert the msg files to python. It's done as part of the build. Your package already has the interfaces declared and will generate them you just have to build it.

Please read through the tutorials at: https://index.ros.org/doc/ros2/Tutorials/

In particular: https://index.ros.org/doc/ros2/Tutorials/Colcon-Tutorial/

If you need more help please open a question on https://answers.ros.org with enough information to reproduce your problem there. We try to keep the trackers for development discussions.

closed 2020-02-11 by tfoote