ros2/rclpy #524: ActionClient

rotu, 2020-03-12:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu Focal
- Installation type:
  - from source
- Version or commit hash:
  - master
- DDS implementation:
  - Fast-RTPS, CycloneDDS
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue
```
colcon test --packages-select test-communication
```

#### Expected behavior
Destroying the action client should happen gracefully.

#### Actual behavior
An error is thrown when the action client or the action server is garbage collected:

http://build.ros2.org/view/Fci/job/Fci__nightly-cyclonedds_ubuntu_focal_amd64/lastBuild/consoleText

```
17: Starting test run test_communication.test_action_client_server__rclpy__rmw_fastrtps_cpp_.launch_tests[Fibonacci]
17: [INFO] [launch]: All log files can be found below /home/buildfarm/.ros/log/2020-03-12-04-03-39-582329-e523ecceac2a-5255
17: [INFO] [launch]: Default logging verbosity is set to INFO
17: test_client_finishes_in_a_finite_amount_of_time[Fibonacci] (test_communication.TestActionClientServer)
17: Test that the second executable terminates after a finite amount of time. ... [INFO] [test_action_server-1]: process started with pid [5258]
17: [INFO] [test_action_client-2]: process started with pid [5260]
17: [test_action_client-2] Sending goal for test number 0
17: [test_action_client-2] Exception ignored in: <function ActionClient.__del__ at 0x7f2115f12700>
17: [test_action_client-2] Traceback (most recent call last):
17: [test_action_client-2]   File "/tmp/ws/install_isolated/rclpy/lib/python3.8/site-packages/rclpy/action/client.py", line 596, in __del__
17: [test_action_client-2]   File "/tmp/ws/install_isolated/rclpy/lib/python3.8/site-packages/rclpy/action/client.py", line 589, in destroy
17: [test_action_client-2]   File "/tmp/ws/install_isolated/rclpy/lib/python3.8/site-packages/rclpy/handle.py", line 124, in __enter__
17: [test_action_client-2]   File "/tmp/ws/install_isolated/rclpy/lib/python3.8/site-packages/rclpy/handle.py", line 106, in _get_capsule
17: [test_action_client-2] rclpy.handle.InvalidHandle: Tried to use a handle that has been destroyed.
17: [INFO] [test_action_client-2]: process has finished cleanly [pid 5260]
17: ok
17: 
17: ----------------------------------------------------------------------
17: Ran 1 test in 1.187s
17: 
17: OK
17: [INFO] [test_action_server-1]: sending signal 'SIGINT' to process[test_action_server-1]
17: [test_action_server-1] Exception ignored in: <function ActionServer.__del__ at 0x7fce84bc3040>
17: [test_action_server-1] Traceback (most recent call last):
17: [test_action_server-1]   File "/tmp/ws/install_isolated/rclpy/lib/python3.8/site-packages/rclpy/action/server.py", line 625, in __del__
17: [test_action_server-1]   File "/tmp/ws/install_isolated/rclpy/lib/python3.8/site-packages/rclpy/action/server.py", line 618, in destroy
17: [test_action_server-1]   File "/tmp/ws/install_isolated/rclpy/lib/python3.8/site-packages/rclpy/handle.py", line 124, in __enter__
17: [test_action_server-1]   File "/tmp/ws/install_isolated/rclpy/lib/python3.8/site-packages/rclpy/handle.py", line 106, in _get_capsule
17: [test_action_server-1] rclpy.handle.InvalidHandle: Tried to use a handle that has been destroyed.
17: [INFO] [test_action_server-1]: process has finished cleanly [pid 5258]
17: test_processes_finished_gracefully[Fibonacci] (test_communication.TestActionClientServerAfterShutdown)
17: Test that both executables finished gracefully. ... ok
17: 
```

#### Additional information

We should probably not be using `__del__` at all, since it executes when the object is garbage-collected, which happens at an indeterminate time. It should probably instead be cleaned up when the parent node terminates.

MatteoRagni, 2020-05-08:
We are experiencing similar problems with Opensplice DDS, on Ubuntu 18.04.

 * Operating System: Ubuntu 18.04.2 LTS (we are planning a test in Windows 10)
 * Installation type: from ROS 2 repository
 * ROS2 Version: Dashing (ros-dashing-ros-base is `0.7.3-1bionic.20200318.045454`)
 * DDS implementation: Opensplice (libopensplice69 version is `6.9.190705+osrf1-1~bionic`)
 * rclpy version: `0.7.10-1bionic.20200318.040148`
 
We are experiencing `segmentation fault` when we try to destroy the server or the client. We tried a post mortem debug of the core dumped, but we have a very short bt with no useful information inside. 

If we do not destroy explicitly, we see the segmentation fault when the stuff is garbage collected at the end of the script.

karl-schulz, 2022-03-30:
Is there any progress on this? We are experiencing the same issue and it is quite misleading for people not knowing about it.