ros2/rclpy #527: Sigint cannot kill python talker run by bash in background mode

Parallel-Hao, 2020-03-24:
## Bug report
I write a bash script to run a python talker program in background, then send sigint to the talker process. The weird thing is that the python talker cannot be killed by sigint. But the cpp talker can be killed.

The same issue was also proposed in https://github.com/ros2/ros2/issues/883. As I suppose this could be a rclpy bug, so I propose it here. 
**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - binaries
- Version or commit hash:
  - foxy
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue

```
# For python test:
python3 /opt/ros/foxy/lib/python3.6/site-packages/demo_nodes_py/topics/talker.py &

# For cpp test:
#../opt/ros/foxy/lib/demo_nodes_cpp/talker &
```
In terminal:
1. bash test.bash
2. find the talker pid, and kill -2 talker_pid

#### Expected behavior
cpp test: talker is killed.
python test: talker is killed.
#### Actual behavior
cpp test: talker is killed.
python test: talker continues running.
#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

hidmic, 2020-04-08:
@Parallel-Hao I'm curious, how did you manage to use Foxy binaries? AFAIK there aren't any yet, we haven't even started with preliminary releases.

Parallel-Hao, 2020-04-09:
@hidmic The foxy version was pulled by:
```
docker pull osrf/ros2:nightly
```

In addition, the above test behaves same in eloquent, which comes from:
```
docker pull osrf/ros:eloquent-desktop
```

hidmic, 2020-04-28:
@Parallel-Hao I'm sorry for the delay. The problem seems to be `python` itself when run from a non-interactive shell (see [here](https://stackoverflow.com/questions/974189/how-to-send-a-sigint-to-python-from-a-bash-script)). Reinstalling `signal.default_int_handler` does the trick.

It's unclear to me whether `rclpy` should override `python`'s default behavior or if it's on client code to take care of it on a case-by-case basis. @wjwwood @clalancette  for feedback.

wjwwood, 2020-06-02:
Sorry for the long delay, I'm still clearing backlogged tabs.

How does that SO link explain this? We register the signal handler in C. They just say you need a python signal handler in the SO thread.

Perhaps that means we're not overriding the signal handler properly from C, or that we need to have a Python signal handler that calls our C signal handler (though IIRC that prevented us from escaping when SIGINT is received inside one of our C extensions, but I don't remember the exact details).

Not being able to SIGINT a python script containing a node because it is in a shell script is not an acceptable user experience in my opinion. I don't think documenting it will be sufficient.

hidmic, 2020-06-02:
> How does that SO link explain this? We register the signal handler in C. They just say you need a python signal handler in the SO thread.

The relevant answer is not the first one, but the second to last. Issue here has nothing to do with `rclpy` signal handlers, but with `bash` and job control. When a process is executed in the background from a non-interactive shell, as is the case above, job control is disabled and thus the default SIGINT handler is a no-op (see `bash` manual [section on signal handling](https://www.gnu.org/software/bash/manual/bash.html#Signals)). That means no `KeyboardInterrupt` is raised, and thus why the process cannot be killed.

> Not being able to SIGINT a python script containing a node because it is in a shell script is not an acceptable user experience in my opinion. I don't think documenting it will be sufficient.

That's fair. `rclpy` signal handling, unlike `rclcpp`'s, relies on `KeyboardInterrupt` to break executors' loops and does not shutdown contexts. We could do the latter, instead of changing shell's behavior. Or better still, we could change both so that SIGINT simply interrupts a spin without shutting down everything (and leave SIGTERM for that purpose).

wjwwood, 2020-06-02:
That sounds better to me, since that’s exactly what rclcpp does.

hidmic, 2020-06-02:
We'll need to sync up `rclcpp` and `rclpy` in this regard then. I'll try to put together a patch, but I don't think it'll make it to Foxy.

hidmic, 2020-10-21:
Alright, IMHO this should get done for Galactic, but I can't say I'll have time to rework `rclpy`'s signal handling to match that of `rclcpp` in the near future. FYI @sloretz @ivanpauno @claireyywang.

ivanpauno, 2020-10-21:
Based on the previous comments, this is how Python behaves by default.

Adding a default signal handler to `rclpy` that raises `KeyboardException` sounds trivial, so I will mark this as `help wanted` (feel free to undo that).

hidmic, 2020-10-21:
> Adding a default signal handler to rclpy that raises KeyboardException sounds trivial, so I will mark this as help wanted (feel free to undo that).

Hmm, IIUC the conclusion of the discussion was: handle SIGINT and shutdown contexts like `rclcpp` does instead of trying to workaround Python's default behavior (i.e. re-installing the default SIGINT handler). In other words, stop relying on `KeyboardInterrupt` to break from spin loops.

ivanpauno, 2020-10-21:
> Hmm, IIUC the conclusion of the discussion was: handle SIGINT and shutdown contexts like `rclcpp` does instead of trying to workaround Python's default behavior (i.e. re-installing the default SIGINT handler). In other words, stop relying on `KeyboardInterrupt` to break from spin loops.

That sounds like a big change, why not just making sure `KeyboardInterrupt` is raised in the signal handler we're installing?

ivanpauno, 2020-10-21:
> That's fair. rclpy signal handling, unlike rclcpp's, relies on KeyboardInterrupt to break executors' loops and does not shutdown contexts. We could do the latter, instead of changing shell's behavior. Or better still, we could change both so that SIGINT simply interrupts a spin without shutting down everything (and use SIGTERM for that purpose).

The behavior in `rclcpp` isn't nice, we're calling `shutdown` asynchronously and that generate random errors when sending SIGINT. I would rather not copy that behavior.

hidmic, 2020-10-21:
> why not just making sure KeyboardInterrupt is raised in the signal handler we're installing?

Having a library silently changing default (and expected) behavior isn't a good idea.

> The behavior in rclcpp isn't nice, we're calling shutdown asynchronously and that generate random errors when sending SIGINT.

That's true. Perhaps the solution is not to `shutdown` but to just wake up loops. Whatever solution we arrive to for one client library is probably extensible to the rest.

ivanpauno, 2020-10-21:
> Having a library silently changing default (and expected) behavior isn't a good idea.

That sounds reasonable.

> That's true. Perhaps the solution is not to shutdown but to just wake up loops. Whatever solution we arrive to for one client library is probably extensible to the rest.

I think that's a reasonable solution to the problem, but it requires a lot of work in both `rclcpp` and `rclpy`.

wjwwood, 2020-10-21:
> That sounds like a big change, why not just making sure KeyboardInterrupt is raised in the signal handler we're installing?

Because it should behave like `rclcpp`?

> The behavior in rclcpp isn't nice, we're calling shutdown asynchronously and that generate random errors when sending SIGINT. I would rather not copy that behavior.

What?

Even if we changed the behavior of entities after shutdown, it wouldn't change how we exit a loop. We'd still use something like `while ok()` and not raising exceptions from where ever we are like `KeyboardInterrupt` in `rclcpp`.

> Having a library silently changing default (and expected) behavior isn't a good idea.

Agreed.

> That's true. Perhaps the solution is not to shutdown but to just wake up loops. Whatever solution we arrive to for one client library is probably extensible to the rest.

I think we should shutdown on SIGINT. That's what `rclcpp` does, that's what ROS 1 did, and I think that's what people expect.

ivanpauno, 2020-10-21:
> What?
> I think we should shutdown on SIGINT. That's what rclcpp does, that's what ROS 1 did, and I think that's what people expect.

I think that imitating the `rclcpp` behavior in `rclpy` is ok.

What I mean is that making that change can generate random errors in places were that didn't happen, and I would rather fix the async shutdown behavior before making a change here.