ros2/rclpy #533: Add default Incompatible QoS callbacks

mm318, 2020-04-01:
## Feature request

#### Feature description

With https://github.com/ros2/rclpy/pull/459 merged in, a user is able to provide a callback for when an participant on the topic is discovered but unable to be matched due to incompatible QoS policies. This feature request is for leveraging this feature even when a user does not provide a custom callback function, by setting a default callback function that prints a warning (initially suggested at https://github.com/ros2/ros2/issues/822#issuecomment-598024481) on the discovery of an incompatible participant on the topic.

#### Implementation considerations

What should the behavior be if the underlying middleware does not support this kind of QoS event callback?

hidmic, 2020-04-08:
@mm318 are you looking forward to implement this feature?

mm318, 2020-04-08:
Hi @hidmic, we already have a pull request opened for this at https://github.com/ros2/rclpy/pull/536.

mm318, 2020-04-15:
Closing this issue as the related pull request has been merged.

closed 2020-04-15 by mm318