ros2/rclpy #549: Subscription segfaults on take when using CycloneDDS

hidmic, 2020-04-27:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - From source
- Version or commit hash:
  - Current [`master`](https://github.com/ros2/rclpy/commit/1c645c873272e1e61b1367d579c59d6367813b9c)
- DDS implementation:
  - ADLINK CycloneDDS, RTI Connext
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue
1. Start up a listener, e.g.
```
ros2 run demo_nodes_py listener
```
2. Start up a talker, e.g.

```
ros2 run demo_nodes_py talker
```

3. Once they are communicating, `Ctrl-C` the talker.

#### Expected behavior
Talker exits and listener stops receiving messages.

#### Actual behavior
Talker exits and listener segfaults.

#### Additional information
Limited backtrace:

```
#0  0x00000000005a8c1d in PyTuple_Pack ()                                                  
#1  0x00007ffff6fd185c in rclpy_take ()                                                    
   from /home/michel/Workspaces/focal_ros2_ws/install/rclpy/lib/python3.8/site-packages/rclpy/_rclpy.cpython-38-x86_64-linux-gnu.so                                                    
#2  0x00000000005f268a in PyCFunction_Call ()                                              
#3  0x00000000005f31fe in _PyObject_MakeTpCall ()
#4  0x000000000056c50e in _PyEval_EvalFrameDefault ()
#5  0x0000000000507a8e in ?? ()
``` 

The issue does not occur when using eProsima Fast-RTPS.

hidmic, 2020-04-27:
Well, it's a regression introduced by https://github.com/ros2/rclpy/pull/542. I'll make a patch.

closed 2020-04-27 by hidmic

wjwwood, 2020-04-28:
Thanks for fixing it @hidmic, @iluetkeb FYI.