ros2/rclpy #564: multi-thread problems in client.call and create_subscription/client/xxx

gong0029, 2020-06-03:
There are two problems for rclpy in multithread enviroment.
(1) Case: when the main thread is blocked at spin and use another thread to execute client.call:

Problem: adding time.sleep(5) between Line 88 and Line 89 
(https://github.com/ros2/rclpy/blob/db1dc7c704d49b90ee7e9cee1af16bfddea4db29/rclpy/rclpy/client.py#L88) will block the function client.call:

Reason: before future.add_done_callback(unblock) is added, the callback function of future has been executed here (https://github.com/ros2/rclpy/blob/db1dc7c704d49b90ee7e9cee1af16bfddea4db29/rclpy/rclpy/executors.py#L349). Then client.call will be blocked at Line 90 "event.wait()".

Solution: I think the threading.Lock is required in client.call and executor. _execute_client.

(2) Case: when the main thread is blocked at spin and create a publisher for a topic use another thread to execute node.create_subscription(client, service, guardcondition has the same problem) for the same topic:

Problem: adding time.sleep(5) between Line 1221 and Line 1222
(https://github.com/ros2/rclpy/blob/db1dc7c704d49b90ee7e9cee1af16bfddea4db29/rclpy/rclpy/node.py#L1221)  will lead to "assert error" here (https://github.com/ros2/rclpy/blob/db1dc7c704d49b90ee7e9cee1af16bfddea4db29/rclpy/rclpy/callback_groups.py#L103)

Reason:  the subscriber has been appended to node.__subscriptions but not appended to callback_group. At this time, the main thread executes to Line 478 (https://github.com/ros2/rclpy/blob/db1dc7c704d49b90ee7e9cee1af16bfddea4db29/rclpy/rclpy/executors.py#L478)  and lead to "assert error" here (https://github.com/ros2/rclpy/blob/db1dc7c704d49b90ee7e9cee1af16bfddea4db29/rclpy/rclpy/callback_groups.py#L103)

Solution:  threading.Lock is required

ivanpauno, 2020-08-27:
@sloretz friendly ping

jacobperron, 2021-09-29:
I'm running into this race, I am trying to create a subscription and client in the main thread while my executor is spinning in a different thread. As a workaround, I tried removing the node from the executor before making any `rclpy.node.Node.create_*` calls or client calls. For example,

```python
executor.remove_node(node)
client = node.create_client(...)
client.call_async(...)
executor.add_node(node)
```

 This seems to work, though it would be nice to have a thread-safe fix in rclpy.